LibreHardwareMonitor/LibreHardwareMonitor #276: Can't control AMD NAVI GPU Fan

tonarii, 2020-06-21:
As mentioned in the title, the Fan speed modification is not working on Navi cards (at least my MSI 5700XT Gaming X). 
But the sensors measurements are ok.
Any idea how i can help you to fix that ?
Thanks .

TimeRaptor, 2020-07-09:
I ended up getting a mod bios. Worked perfectly.

lbibass, 2020-07-09:
What do you mean mod bios? A modded bios that lets Libre control the fan speed?

TimeRaptor, 2020-07-09:
Yeah it's a modded bios. But you can actually control it from the bios itself. I didn't try libre afterwards because I could do it from bios. 
But I would say it should work. When I tried libre before that, it said that PCH fan control wasn't available.(because it was disabled by the manufacturer.)
I can link you to it if you'd like.

Skprrkt, 2020-08-19:
Same problem, Gigabyte 5600 XT Gaming OC.

lbibass, 2020-09-05:
Any updates on this? MSI afterburner can control the fans manually without any blackscreening. (Not sure what changed, maybe there was a change in the new 20.8.3 drivers? I checked the release notes, and didn't see anything)

Hopefully this issue can be resolved soon.

alexxandrv, 2020-11-26:
Have the same problem with Asus ROG Strix Radeon 5700. Radeon software can control GPU fans, but LibreHardwareMonitor cannot.

TwistedKestrel, 2021-01-20:
Have the same issue on an Gigabyte Vega 56. Not sure if it's appropriate to add that to this ticket, but I suspect the problem is not hardware specific and may have something do with the fan controls changing as of version 18.12.2 of AMD's drivers

PhyxionNL, 2021-02-11:
Should be fixed in latest build.

closed 2021-02-11 by PhyxionNL

johny-mnemonic, 2022-04-24:
@PhyxionNL This is unfortunately not fixed as seen in #497