LibreHardwareMonitor/LibreHardwareMonitor #277: ASUS PRIME B350-PLUS (ITE IT8655E/GX)

Gae-IT, 2020-06-21:
The cpu fan speed is not correct, although ITE IT8655E should be supported

could someone fix it?

thanks

osmiyaki, 2020-07-15:
is there a way to help this could be supported?

Phyxion, 2020-07-17:
We'd need someone that has this sensor and knows how to code, I don't have any device with this sensor.

OlympicLarry, 2020-09-30:
@PhyxionNL I have this sensor (Asus Prime 450M-A) and I am a .net programmer (web/sql), but I have no experience with this type of programming. Care to walk me through it?

PhyxionNL, 2020-09-30:
One useful resource is this (as it already implements this sensor for Linux):
https://github.com/a1wong/it87/blob/master/it87.c
There you can see the features it uses:
`FEAT_NEWER_AUTOPWM | FEAT_16BIT_FANS
		  | FEAT_AVCC3 | FEAT_NEW_TEMPMAP | FEAT_SCALING
		  | FEAT_10_9MV_ADC | FEAT_IN7_INTERNAL | FEAT_BANK_SEL
		  | FEAT_SIX_TEMP | FEAT_MMIO`

You will need to look into https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/blob/master/LibreHardwareMonitorLib/Hardware/Motherboard/Lpc/IT87XX.cs and see what's wrong. Just looking at the it87 code the fan registers (FAN_TACHOMETER_EXT_REG) seem to be different for IT8655E, but it might also require some other changes. The easiest way is to check the FEAT_ stuff in it87 and compare with IT87XX.cs and go from there. There's not much documentation on these sensors so it's often trial and error.

oxirt, 2020-11-08:
I'm facing the same issue on ASUS EX-A320M-GAMING motherboard, fan readings are incorrect.

I was able to fix fan readings. In file `IT87XX.cs` line 72-79 for `_hasExtReg` check, I added `chip == Chip.IT8655E`. However, I'm not able to fix PWM controls. Any suggestions?

oxirt, 2020-11-08:
Nevermind, even the PWM control is working now. Will raise a PR shortly.

Gae-IT, 2020-11-09:
> 
> 
> Nevermind, even the PWM control is working now. Will raise a PR shortly.
> 
> Edit: fixed with #327 @Gae-IT I have tested on my board. Can you also test this please?

Hello oxirt,
thanks for your contribution, but for me don'r report correct values
[image]
on the left the results of Asus AI Suite 3

Gae-IT, 2020-11-09:
differences in https://github.com/a1wong/it87/blob/master/it87.c at line 305-306
static const u8 IT87_REG_PWM[] =	{ 0x15, 0x16, 0x17, 0x7f, 0xa7, 0xaf };
static const u8 IT87_REG_PWM_8665[] =	{ 0x15, 0x16, 0x17, 0x1e, 0x1f, 0x92 };

https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/blob/master/LibreHardwareMonitorLib/Hardware/Motherboard/Lpc/IT87XX.cs at line 69-70
                ? new byte[] { 0x15, 0x16, 0x17, 0x1e, 0x1f }
                : new byte[] { 0x15, 0x16, 0x17, 0x7f, 0xa7 };

at line 72:
hasExtReg should be true also for IT8655E ?

it's not easy to comapre both file, it's written in a completely different way (at least for me ;-) )

oxirt, 2020-11-09:
> 
> 
> > Nevermind, even the PWM control is working now. Will raise a PR shortly.
> > Edit: fixed with #327 @Gae-IT I have tested on my board. Can you also test this please?
> 
> Hello oxirt,
> thanks for your contribution, but for me don'r report correct values
> [image]
> on the left the results of Asus AI Suite 3

@Gae-IT My PR is not merged yet, hence you need to pull it, compile it and then test. I hope you're doing the same.

Gae-IT, 2020-11-09:
ops, sorry, I can't compile, I just downloaded latest binaries. I'll wait for the merge, fool sorry :(

oxirt, 2020-11-09:
> it's not easy to comapre both file, it's written in a completely different way (at least for me ;-) )

True, but the overall functionality is same. The sensor has:

- Few *default report registers* to read fan values
- Few *extended report registers* to read fan values
- Few *default control registers* to set fan values
- Few *extended control registers* to read fan values

What is difference between default and extended report/control registers, I'm not sure. My approach was, if defaults are not working, use extended.

oxirt, 2020-11-09:
@Gae-IT if you want to dig deeper, https://github.com/hirschmann/nbfc/wiki/How-to-create-a-NBFC-config has a very good writeup.

oxirt, 2020-11-09:
> I'll wait for the merge

@Gae-IT mods will not merge until you confirm it working :-D @PhyxionNL is there a way around it? Any intrim build that Gae can download?

Gae-IT, 2020-11-10:
I've downloaded binaries from https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/actions/runs/355620077
and also from https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/pull/327/checks?check_run_id=1370547843
but same results, not works for me
[image]

oxirt, 2020-11-10:
[image]

@Gae-IT can you tell me your AMD driver version?

Gae-IT, 2020-11-10:
> 
> 
> [image]
> 
> @Gae-IT can you tell me your AMD driver version?

@oxirt my AMD driver version is the same as you 2.10.13.408
[image]

PhyxionNL, 2020-11-10:
I don't think the driver version matters here. @oxirt Are your fan readings correct with what motherboard reports?

ackstorm23, 2020-12-12:
I have this same problem on my TUF-B450-PLUS-GAMING.

It's IT8655E.

The fan speeds read correctly in another program, Argus Monitor.  However in LibreHardwareMonitor they are way off.

ackstorm23, 2020-12-12:
I used the same links as Gae-IT to test but the fan speeds still report incorrectly.

ackstorm23, 2022-03-30:
This still isn't fixed 2 years later :(

PhyxionNL, 2022-03-30:
> This still isn't fixed 2 years later :(

You're more than welcome to work on it :)

AyamiAlince, 2023-08-04:
> 

3 Years not fixed

closed 2024-08-22 by PhyxionNL

MadGoatHaz, 2024-08-31:
Asus TUF B450m-pro S w' ITE - IT8655e

Issue persists