LibreHardwareMonitor/LibreHardwareMonitor #278: Please add support for Winbond/Nuvoton NCT6687D

PaulPiatek, 2020-06-27:
This chip is used on the MSI MPG Z490 GAMING CARBON WIFI (MS-7C73), but unfortunately not supported by LibreHardwareMonitor yet.

Rem0o, 2020-07-04:
NCT6687D gives the following log:

**LpcIO

Chip ID: Unknown Winbond / Nuvoton / Fintek with ID 0xD592 at 0x4E/0x4F**

So id = 0xD5 and revision is 0x92

Phyxion, 2020-07-21:
There doesn't seem to be much information about this chip, it might be possible to get some basic support for it by adding that identifier / revision, but ideally someone with this chip should add/test it.

fraybentos, 2020-08-10:
If someone can add the NCT6687D identifier and revision and send me the compiled build, I'd be happy to test it and report back. It looks like this chip is used in all the Z490 MSI boards.

fraybentos, 2020-08-15:
This looks request looks like it can be a closed: https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/pull/288

closed 2021-09-29 by PhyxionNL