LibreHardwareMonitor/LibreHardwareMonitor #280: Is there an Option to get LHW running in Powershell?

Just-Ch33s3, 2020-07-18:
I thought I give LHW over OHM a try, does anyone know if i can read out values through a PowerShell script?
Here's my approach so far ( which doesnt seem to work ; ) )

##################################
Add-Type -Path "$CurrentDir\LibreHardwareMonitorLib.dll"
Add-Type -Path "$CurrentDir\HidSharp.dll"

$PC = New-Object LibreHardwareMonitor.Hardware.Computer
$PC.Open()
$IsCpuEnabled = $true
$IsGpuEnabled = $true
$IsMemoryEnabled = $true
$IsMotherboardEnabled = $true
$IsControllerEnabled = $true
$IsNetworkEnabled = $true
$IsStorageEnabled = $true

ForEach ($hw in $PC.Hardware)
	{
		If ($hw.HardwareType -eq "CPU")
		{
			Write-Host "CPU Model:" $hw.Name
			$CPUModel = $hw.Name

			ForEach ($sensor in $hw.Sensors)
			{
				if ($sensor.SensorType -like "*temp*")
				{
				Write-Host "Sensor:" $sensor
				Write-Host "SensorName:" $Sensor.Name
				Write-Host "SensorValue:" $sensor.Value "°C"
				Write-Host "Subhardware Value:" $hw.SubHardware
				Write-Host "----------------------"
				#$CPUTemp = $sensor.Value
				#Write-Host "CPU Load:" $CPULoad "%"
				#$CoreCount++
				}
			}	
		}	
	}	
PC.Dispose()
###################################################
Thanks for your help in advance, have a nice day! 
Greetings!

Phyxion, 2020-07-18:
It should be possible, but you need to call Update on the hardware first, otherwise there are typically no sensors added.

Just-Ch33s3, 2020-07-18:
thanks for your quick reply!
although ive tried $PC.Update instead of $PC.Open it starts to shoot errors like a maniac, im currently searching for a method to properly read out values of sensors .
OHM (Powershell) does the job well but unfortunately it seems that either the Operating System and/or Powershell Build do not correspond well - i get empty values for CPU Temp  back ... the Rest (HDD/GPU/etc.) returns just fine...

Phyxion, 2020-07-21:
Lets continue this in #281 as well.

closed 2020-07-21 by Phyxion