LibreHardwareMonitor/LibreHardwareMonitor #281: (Noob Question) where does the example code go inside Visual Studio?

Just-Ch33s3, 2020-07-18:
Hi again, I've opened up a new Visual Studio Project and would like to create a Console App (Net Framework) to read out values and learn to tweak it... i hope that was correct so far...
As described, I've added both LHMLib + HidSharp to the project, in which "field" does the example code featured on the start page need to go?
[image]
( a simple arrow would probably work wonders ! :) )
Any help is appreciated!
Thanks in advance for your help! :)

Phyxion, 2020-07-21:
You can paste the contents of the example code below "class Program {". Then inside Main method you need to add Monitor(); But it would be best to start with a guide like this:
https://docs.microsoft.com/en-us/dotnet/csharp/programming-guide/inside-a-program/hello-world-your-first-program?tabs=windows

Phyxion, 2020-07-21:
Just closing as it's not an issue with the program, feel free to continue here.

closed 2020-07-21 by Phyxion

cednieb, 2020-07-31:
check this thread : https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/207

Just-Ch33s3, 2020-08-17:
> You can paste the contents of the example code below "class Program {". Then inside Main method you need to add Monitor(); But it would be best to start with a guide like this: [...]

Sorry for the late reply, I'll give it a shot thank you!

Just-Ch33s3, 2020-08-17:
[image]
Unfortunately no luck on this approach, your'e totally right about the getting started page regarding .Net
I'd just love to play some trial and error with this code to test some functions - would you be so kind and check if I did anything wrong here ?

lecciovich, 2022-11-18:
@Just-Ch33s3 I was in same condition as noob C# programmer.
I started from an HelloWorld and retrieved this

`
// LHM Sample Code program
using System;
using LibreHardwareMonitor;
using LibreHardwareMonitor.Hardware;
using HidLibrary;

namespace SampleCode//HelloWorld
{
    class SampleCode
    {
        public class UpdateVisitor : IVisitor
        {
            public void VisitComputer(IComputer computer)
            {
                computer.Traverse(this);
            }
            public void VisitHardware(IHardware hardware)
            {
                hardware.Update();
                foreach (IHardware subHardware in hardware.SubHardware) subHardware.Accept(this);
            }
            public void VisitSensor(ISensor sensor) { }
            public void VisitParameter(IParameter parameter) { }
        }

        static void Main(string[] args)
        {
            Console.WriteLine("Hello World!");

            UpdateVisitor visitor= new UpdateVisitor();

            Computer computer = new Computer
            {
                IsCpuEnabled = true,
                IsGpuEnabled = true,
                IsMemoryEnabled = true,
                IsMotherboardEnabled = true,
                IsControllerEnabled = true,
                IsNetworkEnabled = true,
                IsStorageEnabled = true
            };

            computer.Open();
            computer.Accept(new UpdateVisitor());

            foreach (IHardware hardware in computer.Hardware)
            {
                Console.WriteLine("Hardware: {0}", hardware.Name);

                foreach (IHardware subhardware in hardware.SubHardware)
                {
                    Console.WriteLine("\tSubhardware: {0}", subhardware.Name);

                    foreach (ISensor sensor in subhardware.Sensors)
                    {
                        Console.WriteLine("\t\tSensor: {0}, value: {1}", sensor.Name, sensor.Value);
                    }
                }

                foreach (ISensor sensor in hardware.Sensors)
                {
                    Console.WriteLine("\tSensor: {0}, value: {1}", sensor.Name, sensor.Value);
                }
            }

            computer.Close();

        }

    }
}

`
Hope it can be useful.