LibreHardwareMonitor/LibreHardwareMonitor #282: ASUS EC sensor support

gofspar, 2020-07-19:
Just sponsored the project :D 
was wondering if you could add ASUS EC sensor support for VRM and TMP sensor monitoring.
It's included in HWinfo64.

Ja50n-Kru, 2020-07-19:
Agree - I have been hopeful for this for a while. Unfortunately my skills are not up to the task. This would also cover off issue #260 (and no doubt others)

Phyxion, 2020-07-21:
Yes, it should be possible, but there's not much information about the EC sensor available. I do see there's an ASUS WMI project available (https://github.com/electrified/asus-wmi-sensors) but I think that might duplicate some existing sensors as well. How does HWINFO show it to you?

gofspar, 2020-07-24:
> Yes, it should be possible, but there's not much information about the EC sensor available. I do see there's an ASUS WMI project available (https://github.com/electrified/asus-wmi-sensors) but I think that might duplicate some existing sensors as well. How does HWINFO show it to you?

Just shows up under the sensors tab on HWinfo.
[image]

sqishd, 2020-08-12:
> Yes, it should be possible, but there's not much information about the EC sensor available. I do see there's an ASUS WMI project available (https://github.com/electrified/asus-wmi-sensors) but I think that might duplicate some existing sensors as well. How does HWINFO show it to you?

I believe HWINFO uses the WMI method, would be great if this was added, I have a x370 pro and use the T_Sensor and Water Pump headers that don't show in LHM

HairlessMonkeyBoy, 2020-08-25:
I'm also very interested in this feature. I'd like to start using [FanControl](https://github.com/Rem0o/FanControl.Releases) instead of the terrible AISuite to control my fans, but unfortunately my water T_sensors are not detected.

PhyxionNL, 2020-08-26:
@HairlessMonkeyBoy that would not be possible if this was added as the ASUS WMI doesn't provide write support.

Ja50n-Kru, 2020-08-26:
@PhyxionNL All of the fans are on the standard NCT chip. What we'd be able to do is use the output from the T sensors as an input condition, to drive an effect on the fans controlled by the standard chip. This is exactly what I do with another piece of software that does have the ASUS WMI/EC chip supported.

Ja50n-Kru, 2020-10-24:
@PhyxionNL Is there a view as to whether this is even on the radar, or shall I just move on? Thanks.

PhyxionNL, 2020-10-24:
Someone will need to implement it, and possibly have a system with these sensors so they can test it out.

Ja50n-Kru, 2020-12-23:
Thanks PhyxionNL. Sadly, that is beyond my skills! It does seem the WMI as used here: https://github.com/electrified/asus-wmi-sensors , is the key. I'll keep reading!

berniyh, 2021-02-27:
> Yes, it should be possible, but there's not much information about the EC sensor available. I do see there's an ASUS WMI project available (https://github.com/electrified/asus-wmi-sensors) but I think that might duplicate some existing sensors as well. How does HWINFO show it to you?

I did a little bit of research on this chip last year, which I noted down over here:
https://github.com/electrified/asus-wmi-sensors/issues/62#issuecomment-617375332

Basically, you can read out the sensor using the Linux kernel ec_sys driver, but since you have to do all of the conversions it's a bit tedious.
There is also the a tool for Windows available, which I used for comparisons, see:
https://github.com/hirschmann/nbfc/wiki/EC-probing-tool

Even though you're maybe not using Linux, I was referencing you to that driver, since it is open source and might be helpful when writing something. ;)

xcasxcursex, 2021-06-07:
> It's included in HWinfo64.

They also warn against it for performance reasons and disable it by default. Perhaps this should be done carefully.

axlgundam, 2021-12-06:
I’m wondering if there is anything I can do to get someone to work on this. I have a B550 E Gaming and can see the T_Sensor on both HWInfo and AIDA64. Would love to be able to monitor and set my fan curves based on the T_Sensor using Fan Control.

WanaGo, 2022-02-02:
Argus Monitor uses 2 methods to get T_Sensor, it can use the Asus EC, as well as the Asus ATK. Seems the ATK method is if you have or had AI Suite installed (I did but uninstalled it, but the service remained I think). EC is maybe more unstable (?) I am not really sure.
Couple of screenshots:

[image]

[image]

Wonder if the ATK method is another path to look in to for this? Might be easier the EC ?

ItsMeAubey, 2022-09-21:
Bumping this.. would make my life much, much easier. I can put a small donation towards the completion of this if that would be helpful.

PhyxionNL, 2022-09-23:
This should largely be implemented by @zeule, see here:
https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/blob/master/LibreHardwareMonitorLib/Hardware/Motherboard/Lpc/EC/EmbeddedController.cs

What is your motherboard @ItsMeAubey?

ItsMeAubey, 2022-09-23:
@PhyxionNL 

I've got an Asus Z270e Strix

PhyxionNL, 2022-09-23:
That motherboard would first need to be added to LHM then, maybe @zeule can add this one then :)

zeule, 2023-03-05:
Hi @Svek, @djmeijer,  @mr-stocki, and @416C616E!

You recently contributed to the ASUS Embedded Controller driver in this project. I maintain a similar driver in the Linux kernel, and would like to include data for the motherboards you own. I hereby invite you to submit MR to https://github.com/zeule/asus-ec-sensors, or a patch to the Linux HWMON project, or, should you don't want to do that yourselves, could you please share dumps of the DSDT ACPI table (https://dortania.github.io/Getting-Started-With-ACPI/Manual/dump.html#from-windows) so that I can prepare patches myself? The Linux driver needs additional bit of information from the DSDT, the name of ACPI mutex which guards access to the embedded controller registers so that the monitoring does not clash with the firmware.

jasonlaguidice, 2023-11-26:
@zeule - I know you asked for a file quite some time ago and I don't know if this will help any more but here's a dump from my Asus Z390E... and I'm having some issues with fan control. Hopefully this helps you.

[ASUS Z390E DSDT.zip](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/files/13466485/ASUS.Z390E.DSDT.zip)

Fragrance22600, 2024-06-02:
@zeule - i'd like to add ROG MAXIMUS Z790 DARK HERO support.

Any information you need to add this mobo support, or any way i cna help ?

zeule, 2024-06-02:
> Any information you need to add this mobo support, or any way i cna help ?

If you are sure the board has it (for example, HWINFO shows the EC node), just find the most similar board withing already implemented, copy its definitions and start tinkering with sensors until you find all of them. Alternatively, you can decompile the BIOS file and find sensor addresses there, provided they are displayed by the Q-Fan/Monitor UI. Both ways require access to the hardware, so you have to do that yourselves.

Fragrance22600, 2024-06-02:
for people interested to commit : T_Sensor register is 0x0109 on ROG_MAXIMUS_Z790_DARK_HERO

```
BoardFamily.Intel700, new Dictionary<ECSensor, EmbeddedControllerSource>
 {
     { ECSensor.TempTSensor, new EmbeddedControllerSource("T Sensor", SensorType.Temperature, 0x0109, blank: -40) },
     { ECSensor.TempWaterIn, new EmbeddedControllerSource("Water In", SensorType.Temperature, 0x0100, blank: -40) },
     { ECSensor.TempWaterOut, new EmbeddedControllerSource("Water Out", SensorType.Temperature, 0x0101, blank: -40) },
     { ECSensor.FanWaterFlow, new EmbeddedControllerSource("Water Flow", SensorType.Flow, 0x00be, 2, factor: 1.0f / 42f * 60f) }
 }
```

zeule, 2024-06-03:
> for people interested to commit : T_Sensor register is 0x0109 on ROG_MAXIMUS_Z790_DARK_HERO

Thanks! Could you, please, share ACPI DSDT table dump so I can add this board to the Linux HWMON driver?

Fragrance22600, 2024-06-03:
@zeule 

Decompiled DSDT table is attached.

Hope it will help !

[dsdt.zip](https://github.com/user-attachments/files/15539261/dsdt.zip)

jeffjedi, 2024-07-20:
> for people interested to commit : T_Sensor register is 0x0109 on ROG_MAXIMUS_Z790_DARK_HERO
>
> ```
> BoardFamily.Intel700, new Dictionary<ECSensor, EmbeddedControllerSource>
>  {
>      { [...]

This "T_Sensor" register works for the regular ROG MAXIMUS Z790 HERO as well (my motherboard).
I added the sample output in my pull request while using 3 temp sensors inside a custom water loop and the reading checks out as being the same values displayed within HWiNFO and Armoury Crate (as well as consistent with environment temperature).
Can you please bump the pull request? (#1375)
I use my compiled dll with FanControl, but I have to keep manually copying the file each time they update the tool.

Thanks.

teodorstan, 2024-08-25:
Related:
- Issue: https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/1439
- PR: https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/pull/1440

jeffjedi, 2024-08-26:
I have a question for @zeule or anybody that could possibly help me debug the other sensors.

Is there a way to determine the hardware register for this VRM sensor (which provides the actual VRM temperature) that HWiNFO64 displays?
For my MB (MAXIMUS Z790 HERO) the current sensor with the string "VRM" (class SuperIOHardware) is actually the MB's CPU sensor (of course I'll triple test it before adding any new PR).

[image]

As you see in the image above the actual VRM temperature is within that EC section together with three pin sensors available on the MB.
I also would like to bring the CPU_OPT fan RPM reading too if there's a way to obtaing the correct address of that sensor.
Currently I'm trying to bring the Flow Rate (in my case a water cooling flow sensor) to LibreHardwareMonitor using the register '0x00b0' (instead of '0x00be') because it's the only register I been getting a number as result, which is the integer portion of what HWiNFO64 displays in the image above as well, but despite it being the integer of that ("6" in the image instance) I think the HWiNFO64 got this reading as double the actual value of water being displaced (as in my BIOS it would display 3 too, but in it it always shows int as well) because using the value 6 and my pump running at 20% its speed it would mean 1.800 liters per hour, which clearly is not correct, as the declared capacity is 1.100 liters maximum without obstacles, 900 liters (half) sounds more realistic.

I'm using Windows 11 Pro so I don't know of any tools that list the hardware's sensors.
I even tried calling the Asus class (ASUSHW) using WMI but it replies as non existing class.

Thank you.

zeule, 2024-08-26:
> Is there a way to determine the hardware register for this VRM sensor (which provides the actual VRM temperature) that HWiNFO64 displays? 

Trial and error or decompiling  the BIOS.

> Currently I'm trying to bring the Flow Rate (in my case a water cooling flow sensor) to LibreHardwareMonitor using the register '0x00b0' (instead of '0x00be') because it's the only register I been [...]

Presumably, the reading for the flow sensors is in RPMs.

mirh, 2024-08-26:
I thought #894 did have the VRM reading?
Anyhow, besides the bios you can also try looking into gamefirst or AI Suite (whatever the OEM software name is) with a decompiler

jeffjedi, 2024-08-27:
> Trial and error or decompiling the BIOS.

Thank you, I'm going about it using the trial/error method hoping only reading the registers doesn't crash Windows or the MB.

> Presumably, the reading for the flow sensors is in RPMs.

In the Maximus Hero there's a separated 3-pin header exclusively for the pupose of water cooler flow:
[image]

Works like a FAN header but provides that 12V without changing the voltage to control the speed based on reading, that 12V is exclusively to power the sensor itself, and the data wire (labeled "W FLOW IN" in the image) receives the "revolutions" count value that the sensor provides. The header is entirely separade from the pump PWM header as well, and it's 100% used for reading only, the MB doesn't have any related automation or logic in regards to the PWM signal sent to the water pump.
It should output a float value (like displayed in HWiNFO) but both on BIOS or ASUS software it only displays the integer value, but corretly as 3 in that case I mentioned.

> I thought #894 did have the VRM reading?

The code, although working, is not correct, the real **VRM** reading is within **EC registers** (as displayed in my HWiNFO image).
I know this because the value labeled as VRM in the current code is actually the **CPU** value using the MB sensor (which would be the MB equivalency of **CPU core tempereture**). The VRM temperature is usually higher than the CPU reading while idle and the values during load are consistent with MB CPU sensor as well (usually 2 to 5 degrees lower than **CPU Package** sensor). The sensor CPU Package is correct in the current code.
According to my testing inside the code there's a **Chipset** string that never comes back up to the UI or software importing the library because the Chipset string is assigned to **index 12** and the Chipset sensor is actually **index 13** (comes up to UI as **Temperature '#'13**).
Please remember that **all this information is only valid for ASUS MAXIMUS Z790 HERO**, I can't evaluate if it's the same in other MB models or not.

zeule, 2024-08-27:
> In the Maximus Hero there's a separated 3-pin header exclusively for the pupose of water cooler flow:….

It's the same in other boards.

> It should output a float value (like displayed in HWiNFO) but both on BIOS or ASUS software it only displays the integer value, but corretly as 3 in that case I mentioned.

It simply can't, for instance because calibration of sensors varies:

https://shop.alphacool.com/en/shop/controllers-and-sensors/flow-sensor/sen-digmesa-flow-sensor-g1/4-5.6mm-without-cable
https://download.alphacool.com/datasheet/ENG_17558_Alphacool_ES_flow_and_temperature_indicator_HighFlow_with_RPM-Signal_datasheet.pdf
https://download.alphacool.com/legacy/1017207_rpm_to_flow_rate_overview.pdf