LibreHardwareMonitor/LibreHardwareMonitor #283: ITE IT879XE control support

AdriaanDeVos, 2020-07-22:
**Problem description**
As a user I would like to be able to control the fans of an IT879XE. The current application provides the actual values of this device, but the fan control functionality does not seem to work.

**Describe the solution you'd like**
Expand current read-only support of IT879XE to support full fan control.

**Describe alternatives you've considered**
Check this project for information about IT879XE support. https://github.com/openhardwaremonitor/openhardwaremonitor/commit/7bf4f655d2c3ad9b660b6703c632b1d5694778ca

**Additional Context**
Hardware used:
Gigabyte X570 Aorus Pro Rev 1.0
ITE IT8688E for 6 temperature readings and 5 fans, ITE IT879XE for 3 more temperatures and 3 more fans

PhyxionNL, 2020-08-26:
Fan control is broken in OHM as well, someone with this chip needs to try and fix this and submit a PR for it.

DarkIrata, 2020-09-24:
https://github.com/openhardwaremonitor/openhardwaremonitor/issues/1287 
An old open issue to that problem. Just to link it.

dephcon, 2020-11-18:
also have this issue, however my x570 Master rev1.0 has a IT8975E as the secondary fan controller with the primary IT8688E.

CreeT6, 2020-12-07:
i have x570 aorus pro, same issue

virusmoere, 2021-01-10:
Is there  anything I can do to support development in this case? I could supply remote access to the sensor if needed for some testing.

ajhix36, 2021-11-26:
Any work on this? I have the same issue :(

Koli0842, 2022-05-02:
I also encountered this issue. RPMs are read just fine, but control does not seem to do anything (from FanControl). How does one figure this out without breaking things? If the Linux WMI support gives a good reference, I'm willing to test this on my X570 Pro, but would rather not drill into it headfirst.

NormanPCN, 2022-10-03:
I am in this same boat. Gigabyte z390 Aorus Master MB.

it879xe fan headers are reading speed but no control (FanControl software).  No control or the MB and FanControl are fighting over control. Force applying in FanControl is no change.

I can likely move my fans to other fan headers. it8688e controller. I have only one fan on the it879xe right now.

PhyxionNL, 2023-03-21:
Continue at #251.

closed 2023-03-21 by PhyxionNL