LibreHardwareMonitor/LibreHardwareMonitor #294: IT8728F control

XNtricity, 2020-08-27:
**Problem description**
Systems with the IT8728F controller operate in read-only mode. The application can see the controller and its values, but fans connected cannot be controlled.

**Describe the solution you'd like**
Update software to allow user control of fans connected to the IT8728F controller.

**Describe alternatives you've considered**
SpeedFan supports this controller, however SpeedFan is old and no longer supported. I would like to migrate to a newer and supported application.

**Additional Context**
Hardware used:
Gigabyte Z87X-UD3H-CF
Motherboard has 6 fan headers: CPU_FAN, CPU_OPT, SYS_FAN1/2/3, and SYS-FAN4 is 3-pin only.

DiegoJp, 2020-11-14:
Second this, my motherboard (Z77X-UD3H) also has ITE IT8728F.
Programs using this library can only show readings but no controlling.
SpeedFan and several other programs/libraries report 5 fan speed and 3 controls.
SpeedFan can control all of them.

axcester, 2021-01-03:
Bump. My Gigabyte Z87X-UD4H also has this controller

LAzinheira, 2021-01-25:
Same issue with GA-78LMT-USB3 (rev5.0) using the same controller.
Speedfan allows control of fan speed.

PR #400 (2021-03-12): Add fan control for IT8728F
Add support for controlling fans on the IT8728F controller, mostly used on Gigabyte 7/8 mainboards.
Fixes #294.

Hope this is the only change needed, but it seems so from previous commits and works on my machine.
closed 2021-03-12 by PhyxionNL