LibreHardwareMonitor/LibreHardwareMonitor #297: Cant parse BIOS datetime

Teriand, 2020-09-08:
subj
```
2020-09-08 11:21:20.9663|INFO|OhmGraphite.Program|Prometheus port: 4445
2020-09-08 11:21:21.0818|INFO|Topshelf.HostFactory|Configuration Result:
[Success] Name OhmGraphite
[Success] DisplayName Ohm Graphite
[Success] Description Extract hardware sensor data and exports it to a given host and port in a graphite compatible format
[Success] ServiceName OhmGraphite
2020-09-08 11:21:21.0928|INFO|Topshelf.HostConfigurators.HostConfiguratorImpl|Topshelf v4.2.1.215, .NET Framework v4.0.30319.42000
2020-09-08 11:21:21.1913|ERROR|OhmGraphite.PrometheusServer|Exception: starting prometheus server!System.ArgumentOutOfRangeException: Параметры Year, Month и Day описывают непредставимое значение DateTime.
   в System.DateTime.DateToTicks(Int32 year, Int32 month, Int32 day)
   в LibreHardwareMonitor.Hardware.BiosInformation.ParseBiosDate(String biosDate)
   в LibreHardwareMonitor.Hardware.SMBios..ctor()
   в LibreHardwareMonitor.Hardware.Computer.Open()
   в OhmGraphite.PrometheusServer.<Start>b__4_0()
   в OhmGraphite.LoggerUtils.LogAction(Logger logger, String msg, Action action)
```
and screen from aida+cpu-z

[image]

rmattuschka, 2020-09-08:
your BIOS returns the date in a wrong format dd/mm/yyyy.
mm/dd/yyyy is requirement!

Teriand, 2020-09-09:
U can add check wrong format and dont ex ?

PR #432 (2021-04-27): Add check and swap wrong bios date format
Fixes #297
closed 2021-04-27 by PhyxionNL