LibreHardwareMonitor/LibreHardwareMonitor #303: Publish LibreHardwareMonitorLib nuget

MoshiMoshi0, 2020-09-29:
It would be nice if we had appveyor builds of LibreHardwareMonitorLib on nuget so we dont have to download/clone the whole project.
Unfortunately I never published anything to nuget so I cant help much, but looking at this it looks simple enough: 
https://www.appveyor.com/docs/deployment/nuget/

PhyxionNL, 2020-10-02:
Yeah, I agree, this is useful. Nupkg is already built, it needs to be published. Ideally I'd like to move away from AppVeyor and to GitHub Actions but couldn't manage to get it to build there. Although I've not spent much time on that.

PhyxionNL, 2020-10-09:
Fixed with https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/commit/7ac7c29013c6292a43d14d4da34ef5845b11a13b.

https://www.nuget.org/packages/LibreHardwareMonitorLib/

closed 2020-10-09 by PhyxionNL