LibreHardwareMonitor/LibreHardwareMonitor #310: Support AMD family 17h processors (Zen 3)

DevTechProfile, 2020-10-05:
Hi, I'm a dev from the CapFrameX team. We have a similar hardware lib. Any plans to support Zen 3? 

The only source I have is a Linux patch: https://git.kernel.org/pub/scm/linux/kernel/git/groeck/linux-staging.git/commit/?h=hwmon-next&id=2835d860d3fcc3e4829e96987544e811d35dee48

PhyxionNL, 2020-10-09:
Sure, but let's wait until they're released so it can be tested.

DevTechProfile, 2020-10-10:
We have implemented an experimental Zen 3 support: https://github.com/CXWorld/CapFrameX/commit/9b55c917d59240e6278b2e7437c2500fb74559f9

Patrick Schur checked the constants in the latest AGESA supporting Zen 3.

PhyxionNL, 2020-10-10:
Nice work, please submit a PR, I'll gladly review it. Some of your other changes (e.g. to GPUs) are welcome too.

luctowers, 2020-11-10:
Hi, I need this for a project I'm working on. I have a ryzen 7 5800x to test. I can test any changes put up. Or I can try my hand at adapting @DevTechProfile's changes myself if that's ok.

PhyxionNL, 2020-11-10:
> Hi, I need this for a project I'm working on. I have a ryzen 7 5800x to test. I can test any changes put up. Or I can try my hand at adapting @DevTechProfile's changes myself if that's ok.

Yeah, sure, either should be fine, thanks 😁

cwbr, 2020-11-10:
I could also provide testing on a 5900x (2 CCDs).

luctowers, 2020-11-10:
> I could also provide testing on a 5900x (2 CCDs).

Sweet, tomorrow is a stat holiday in Canada. So, I'll try and hack a build together.

luctowers, 2020-11-12:
Hmm, I may be in over my head. I got tCtl, tDie, core voltage and soc voltage working only because they were extermely similar Zen 2.  But, per-core clocks and VID have defeated me so far will keep digging.

PhyxionNL, 2020-11-12:
Have you checked this: https://github.com/CXWorld/CapFrameX/commit/9b55c917d59240e6278b2e7437c2500fb74559f9
?

DevTechProfile, 2020-11-12:
Per-core clocks work but we haven't VID yet.

PR #328 (2020-11-14): Add support for AMD Ryzen 5000 processors
resolves #310 (may not work for all future zen 3 chips, but should good for ryzen 5000)

Zen 3 seems to much be pretty much the same as Zen2 sensor-wise. Just a new a family and model.

I discovered a couple oddities while exploring and testing though.

1. Calculation of frequencies is currently done with a hard-coded base clock of 100MHz. We should be using the sensed base-clock, 99.8MHz on my system. We should create a seperate issue for that.
2. While per-core frequencies/VID/multipliers seem mostly accurate, it seems that the way we are currently s [... cut]
closed 2020-11-14 by PhyxionNL