LibreHardwareMonitor/LibreHardwareMonitor #313: Bug with NCT6791D: RPM readout update for zero RPM missing

Korkman, 2020-10-14:
Asus X99-E Mainboard with Nuvoton NCT6791D: The RPM readout of my three installed fans stops updating on the last non-zero readout when setting the fans to zero RPM. The effect being that those fans never get to display zero RPM.

Korkman, 2020-10-14:
When set to zero RPM, the fans disappear from the list after restarting LHM. Showing "hidden sensors" does not show them either. Starting them again, producing non-zero RPM, makes them appear again in the running LHM instance.

Korkman, 2020-10-18:
So in some way it does know the number of the fan but skips those reporting zero RPM for some reason:
[image]

This must be filtered somewhere, but I don't see it in the code yet. Other software reports all six fans.

PhyxionNL, 2020-10-20:
Thanks, it's merged.

closed 2020-10-20 by PhyxionNL