LibreHardwareMonitor/LibreHardwareMonitor #322: Ubutu 18.04 LTS usage of .lib results in Error when starting device: System.IO.FileNotFoundException: Could not load file or assembly 'Mono.Posix, Version=2.0.0.0, Culture=neutral, PublicKeyToken=0738eb9f132ed756'. The system cannot find the file specified.

ManuelKugelmann, 2020-10-29:
Used the .nuget in a .Net Core app.
When I run the app on Ubuntu 18.04 LTS I get:

Error when starting device: System.IO.FileNotFoundException: Could not load file or assembly 'Mono.Posix, Version=2.0.0.0, Culture=neutral, PublicKeyToken=0738eb9f132ed756'. The system cannot find the file specified.

Putting the Mono.Posix.dll into the app folder does not resolve the problem. The Version, Culture and PublicKeyToken of the .dll match.

PhyxionNL, 2020-10-29:
If you find a fix, please submit a PR for it. At the moment it's only tested on Windows.

gareth7562, 2021-02-05:
Works in manjaro with .net core 5 and the following nuget package https://www.nuget.org/packages/Mono.Posix-4.5/

charlie-haley, 2021-04-04:
Im having the same issue on Arch Linux. Installing `Mono.Posix-4.5` didn't help either. This is just using LibreHardwareMonitorLib in a net5.0 console app

Azaran, 2022-01-26:
I had to add the mentioned `Mono.Posix-4.5` Nuget package and install `mono` on host machine. I was using CentOS 7.