LibreHardwareMonitor/LibreHardwareMonitor #330: CPU sensors are reading from wrong cores due to CPU affinity issue

luctowers, 2020-11-13:
Amd Zen relies on the thread executing on the respective core in order to get per-core sensor data. This does not seem to be working at the moment.

For example, in this screenshot a multitude of cores are shown at 4850MHz when in reality only one of them is.

[image]

luctowers, 2020-11-13:
Byproduct of #328

PhyxionNL, 2020-11-20:
Can you tackle this @luctowers? I do not see this issue on my CPU.

luctowers, 2020-11-21:
> Can you tackle this @luctowers? I do not see this issue on my CPU.

Will look into it this weekend, maybe it's something screwy with my environment.

PhyxionNL, 2021-02-08:
I think this might be fixed in latest build, can you check it @luctowers.

closed 2024-01-29 by luctowers