LibreHardwareMonitor/LibreHardwareMonitor #335: More sensors for Nuvoton NCT6798D

holgerpieta, 2020-11-18:
The Nuvoton NCT6798D on my Asus TUF Gaming B550M-Plus (Wi-Fi) appears to be supporting more temperature sensors than are currently read out in LHM.
They are visible both in BIOS and in HWiNFO and show reasonable values, so they seem to be really there and not just unconnected ports. Which is a bit of a shame, because the "CPU temperature" used by the BIOS to control the fans is one of the sensors missing.

I had a look at how support for the NCT6798D was added in PR #79 and it looks like it was just enabled as a supported sensor in the generic NCT677X support. Maybe we need a new support for NCT679X chips with at least 3 more temperature sensors.
Also I'm a bit surprised to see 7 temperature sensors configured in NCT677X.cs, but only 6 are visible (none hidden) when I run the program. Is there some other part in the code that removes sensors again if the values are bogus or something like that?

I'm willing to dig into that, but I'm missing some fundamental build instructions. I gathered I need Visual Studio. VSCode is not enough?

PhyxionNL, 2020-11-20:
You need Visual Studio 2019, you can download the free community edition.

Ja50n-Kru, 2020-11-20:
You may well find that the extra sensors you are seeing are actually under the 'ASUS EC' controller, which is currently not supported.

holgerpieta, 2020-11-20:
OK, I'll see if I get it to compile and then try some changes.

Looking at issue #282 I see that the sensors are displayed in HWiNFO in the section "ASUS EC". However, the sensors I'm missing are listed under "NCT798D", so I'm quite optimistic that they can be read out from there.

holgerpieta, 2020-11-20:
Got it to build in VScode without significant problems. Took just 6 seconds, which was so fast, I spend a couple of minutes hunting for errors before I found the working binary in ./bin.
Will look into the additional sensors tomorrow.

PR #339 (2020-12-07): Fixed temperatures for NCT6796D... and added ASUS TUF GAMING B550M-PLUS (WI-FI)
Updated temperature readings for NCT6796D, NCT6796DR, NCT6797D and NCT6798D based on the available documentation for NCT6796D.
No documentation could be found for the later models, so I assumed they are similar.

Also added ASUS TUF GAMING B550M-PLUS (WI-FI) to the list of supported mainboards.

I'm not yet happy with the names for the temperatures, e.g. PECI appears to be an Intel thing and not AMD. But to fix the names more info is needed, about which temperature source is measuring what.
The badly named PECI see [... cut]
closed 2020-12-07 by PhyxionNL