LibreHardwareMonitor/LibreHardwareMonitor #336: LibreHardwareMonitorLib NuGet does not get published correctly

MoshiMoshi0, 2020-11-19:
I noticed that on each ci build the version is set to `0.8.5`, this causes `rohith/publish-nuget` to not publish the new nuget because there is a [existing version check](https://github.com/brandedoutcast/publish-nuget/blob/c12b8546b67672ee38ac87bea491ac94a587f7cc/index.js#L108).

The version should be updated to include the ci build number, so for example `0.8.5-ci39`. I guess it should also be marked as prerelease. Ideally the dll would also include the ci version as the file version but that might be harder to do and unnecessary.

PhyxionNL, 2020-11-20:
I do not think that a "normal" NuGet package should be uploaded after any commit but rather when enough useful changes have been made and then should the version number be increased. I've added the version to a Directory.Build.props now to make it a bit easier to bump the version. Prerelease versions could be useful though, if you'd like that, please submit a PR for that.

MoshiMoshi0, 2020-11-20:
Yes, not on each commit. I think best solution is to push prerelease versions on each merged PR and stable versions on each new release. You could I guess bump the versions manually but I think that is doing the same work as auto versioning with ci build number. If someone adds a new sensor with a PR it will be automatically included in a new ci build, otherwise you would have to remember to bump the version and do a new release.

I've read that nuget automatically marks versions as prerelease when you add a suffix to the version, like `-beta`, `-alpha`, `-build42` etc. so you would only really need to detect if its a PR build and add a ci build number to the version.
I'll try to read up more about github actions as I never used them yet, and maybe do a PR but I dont want to push stuff blindly.

PhyxionNL, 2020-11-20:
You can fork it and try GitHub actions out until you arrive at a working configuration, but I suspect that also the project's versioning needs to be changed.

PR #338 (2021-01-10): Add CI build number to project version when PR is merged
Updates the `Directory.Build.props` version with a ci build number when a PR is merged, this will publish a new version of LibreHardwareMonitorLib to nuget that will be marked as prerelease. Don't know if this is a "proper" solution so feel free to close this, I dont really want to touch github actions again.

Also cleaned up artifact upload for net452, netstandard20 and nupkg.
Fixes #336.

As a side note I think github actions are terrible, they try to combine code in yaml with so many random restrictions that it becomes a chore to do a [... cut]
closed 2021-01-10 by PhyxionNL