LibreHardwareMonitor/LibreHardwareMonitor #346: LibreHardwareMonitorLib.sys detected as Trojan.BitCoinMiner

zapper83, 2020-12-17:
I'm using [Sidebar Diagnostics v3.5.6](https://github.com/ArcadeRenegade/SidebarDiagnostics) which utilizes LibreHardwareMonitor and today Malwarebytes did an auto-scan and detected LibreHardwareMonitorLib.sys as Trojan.BitCoinMiner. I'm going to bet this is a false-positive but i would like a confirmation please.
I posted this issue in the Sidebar Diagnostics repo and the dev told me to post here.

-System Information-
OS: Windows 10 (Build 18362.1198)
CPU: x64
File System: NTFS
User: System

-Blocked Malware Details-
File: 1
Trojan.BitCoinMiner, C:\Users\xxxx\AppData\Local\SidebarDiagnostics\app-3.5.6\LibreHardwareMonitorLib.sys, Quarantined, 945, 886905, 1.0.34027, , ame, , 0C0195C48B6B8582FA6F6373032118DA, 11BD2C9F9E2397C9A16E0990E4ED2CF0679498FE0FD418A3DFDAC60B5C160EE5

PhyxionNL, 2021-02-08:
It's a false positive obviously, cannot do anything about that, so closing.

closed 2021-02-08 by PhyxionNL

idkman2021, 2022-09-07:
what is the point of asking for confirmation, do you think if there was a miner, the devs would be like "ooh yes our app has a bitcoin miner"?