LibreHardwareMonitor/LibreHardwareMonitor #351: Cannot access coolant temperature from Corsair H115i.

mmmboost, 2020-12-23:
It would be extremely useful to be able to access coolant temperatures from things like all-in-one closed-loop coolers.  I current have a Corsair H115i and I despise using the Corsair software to monitor and control my fans.  The coolant temp, pump speed, and fan speed all show up in HWinfo64.

GrumpyNyanCat, 2021-02-14:
Hi, great work here. I'm using the Corsair H80igt and can also see the sensor temps of the water in HWinfo64. Would really be useful, if it would be possible to see them also here. I want to get rid of the corsair software and this would be the best way.

Chumba, 2021-06-24:
I've got an H100i and I'd love to be able to control its fans and see temps, I took a look at the code here and I'm afraid its above my head so I'm not sure how much I can help from that side. I did do some research however; the USB protocol is detailed here, and based on the code for the Corsair PSUs it doesnt seem it should be too different of an implementation. What do you guys think?

https://forum.corsair.com/forums/showthread.php?p=673781

ezfox, 2022-03-29:
Same problem here...has there been any progress for this already?

Iruberiam, 2022-08-14:
I'd also love this feature. Is it at all possible?

OrbitalCannon, 2023-05-02:
Would love to be able to read for the H150i as well.