LibreHardwareMonitor/LibreHardwareMonitor #354: Please put binaries in Releases

lucknaumann, 2020-12-29:
Thanks for this great program! I would like to ask you to please put binaries for download back in the Releases section as trying to figure out the Actions section is too confusing and time-consuming. It's much simpler to go directly to the Releases and download. Thank you.

oldium, 2020-12-30:
I can see `binaries.zip` in the Releases section.

lucknaumann, 2020-12-30:
Thanks for your response. Yes, there are binaries in the Releases section but I am referring to what appears to be a change in this policy. From the README:

> Do I have to compile the program myself?
> You can download the latest builds [here](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/actions). Click on a result and download Binaries under Artifacts.

The link leads to the Actions section, **not** the Releases section:
https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/actions
So, unless I have misunderstood this statement, this is what I'm referring to. If I have misunderstood, then ignore this request and I'll close it. Thanks.

closed 2021-01-10 by PhyxionNL

lucknaumann, 2021-01-10:
@PhyxionNL Thanks for the update clarification. 😃

commit ca09e88e (2021-01-10): Update README.md

Fixes #354.