LibreHardwareMonitor/LibreHardwareMonitor #355: Bad firmware version scrape in KrakenX3.cs?

konistehrad, 2021-01-05:
https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/blob/1506006f4ec1e64096307554a63a57106cbb53c7/LibreHardwareMonitorLib/Hardware/Controller/Nzxt/KrakenX3.cs#L55

Rather than `FirmwareVersion = $"{_rawData[0x11]}.{_rawData[0x11]}.{_rawData[0x13]}";` I believe that second `0x11` should be `0x12`. I'm working on adding support for Nzxt's Smart Device V2, and it shares an initialization routine with the Kraken. I'd open a PR for this, but I don't have hardware to ensure that this is correct.  ([liquidctl seems to indicate](https://github.com/liquidctl/liquidctl/blob/59f2b7c1db9a676480894209b95263340ed26365/liquidctl/driver/kraken3.py#L178) that this is correct, however.)

CZonin, 2021-01-06:
I think this could be related to an issue I opened awhile ago https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/345

PhyxionNL, 2021-01-10:
Is likely x12 indeed, reading x11 twice doesn't make sense.

commit ae60dcec (2021-01-10): Update KrakenX3.cs

Fixes #355.
closed 2021-01-10 by PhyxionNL