LibreHardwareMonitor/LibreHardwareMonitor #356: Only one fan reporting rpm ASRock Z490 Aqua - Nuvoton NCT6796D-R

iBeta666, 2021-01-06:
Hello LHWMonitor,

I could not get fan speeds using the dll, so tried with the app, and it only shows the rpm for one fan. There are two other fans running which do not get reported. 

[image]

Thanks.

iBeta666, 2021-01-14:
Not sure even the Sensor chip reported is correct. According to HWInfo64, which reports all running fans correctly, the chip nos. are:
Nuvoton NCT6683D (chassis 1)
Nuvoton NCT5585D (chassis 4/6)

6d6178667269747a, 2022-05-27:
I have the same problem on my Asrock H570M-ITX/ac. HWINFO64 detects a Nuvoton NCT5585D, but librehardwaremonitor detects a NCT6796D-R. NCT5585D is actually true, i checked the chip label.

samaelestevez, 2022-10-11:
I have the same issue with an Asrock Z690i. The chip is Nuvoton NCT6683D