LibreHardwareMonitor/LibreHardwareMonitor #361: MSI B550 Gaming Carbon WIFI // T_SENS1 Temperature missing

FocalDE, 2021-01-18:
Hey All, 
Id kindly like to ask, if someone can have a look into the missing T_SEN1 temperature for the MSI B550 Gaming Carbon WiFi Mainboard, allthough everything else seems to be supported just fine. It would be awesome if this reading could be integrated.

Background: This MB has a temperature sensor socket (T_SEN1), which can be used to measure the temp with a probe deliverd with the MB at a spot of the users choice. Id like to use this probe and the T_SEN1 temperature reading to control fan speeds with the tool FanControl (watercooling).
T_SEN1 is being shown in HWiNFO just fine (NUC126; T_SEN1).
Id be happy to test any results if required.
Thank you a million times.

Dippndots, 2021-02-08:
I have the same issue on my MSI X399 MEG Creation, so hopefully if this is addressed, it ends up covering all MSI boards

FocalDE, 2021-02-09:
Glad to see i am not the only one with this requirement :)

Alcolawl, 2024-01-06:
Commenting for visibility. I too would like this sensor to be able to be monitored. Looks like there are some other open issues for T_Sensors on other motherboard brands as well. If any information is needed by the devs to get this done, please let me know.