LibreHardwareMonitor/LibreHardwareMonitor #363: Temperatures and sub-hardware showing up in interface but not when using the code

NaturalBornCamper, 2021-01-21:
I know this might be outside the scope of C# and not really an issue, but when using the interface, I have all these sub hardware and temperature sensor values:
[image]

However, when using the `LibreHardwareMonitorLib.dll` file:

- My motherboard doesn't show any sensors or sub-hardware
- My CPU's temperature values are all `null` (`None` in Python)

I even thought that maybe the temperatures were in the array `sensor.Values` instead of `sensor.Value` but the list is always empty. Here is the output of the Motherboard and CPU sensors/sub-hardware/sub-hardware's sensors:
```
HARDWARE TYPE: Motherboard, NAME: Gigabyte Z390 AORUS PRO WIFI-CF (0 sensor(s), 0 sub hardware(s)) 

HARDWARE TYPE: Cpu, NAME: Intel Core i9-9900K (36 sensor(s), 0 sub hardware(s)) 
Sensors: 
Type: Load, Id: 1, Name: CPU Core #1, Value: <class 'float'> 21.15384 %, len(sensor.Values): 0 
Type: Load, Id: 2, Name: CPU Core #2, Value: <class 'float'> 19.23077 %, len(sensor.Values): 0 
Type: Load, Id: 3, Name: CPU Core #3, Value: <class 'float'> 9.615385 %, len(sensor.Values): 0 
Type: Load, Id: 4, Name: CPU Core #4, Value: <class 'float'> 5.769229 %, len(sensor.Values): 0 
Type: Load, Id: 5, Name: CPU Core #5, Value: <class 'float'> 13.46154 %, len(sensor.Values): 0 
Type: Load, Id: 6, Name: CPU Core #6, Value: <class 'float'> 13.46154 %, len(sensor.Values): 0 
Type: Load, Id: 7, Name: CPU Core #7, Value: <class 'float'> 7.692313 %, len(sensor.Values): 0 
Type: Load, Id: 8, Name: CPU Core #8, Value: <class 'float'> 11.53846 %, len(sensor.Values): 0 
Type: Load, Id: 0, Name: CPU Total, Value: <class 'float'> 12.74039 %, len(sensor.Values): 0 
Type: Temperature, Id: 0, Name: CPU Core #1, Value: <class 'NoneType'> None °C, len(sensor.Values): 0 
Type: Temperature, Id: 1, Name: CPU Core #2, Value: <class 'NoneType'> None °C, len(sensor.Values): 0 
Type: Temperature, Id: 2, Name: CPU Core #3, Value: <class 'NoneType'> None °C, len(sensor.Values): 0 
Type: Temperature, Id: 3, Name: CPU Core #4, Value: <class 'NoneType'> None °C, len(sensor.Values): 0 
Type: Temperature, Id: 4, Name: CPU Core #5, Value: <class 'NoneType'> None °C, len(sensor.Values): 0 
Type: Temperature, Id: 5, Name: CPU Core #6, Value: <class 'NoneType'> None °C, len(sensor.Values): 0 
Type: Temperature, Id: 6, Name: CPU Core #7, Value: <class 'NoneType'> None °C, len(sensor.Values): 0 
Type: Temperature, Id: 7, Name: CPU Core #8, Value: <class 'NoneType'> None °C, len(sensor.Values): 0 
Type: Temperature, Id: 8, Name: CPU Package, Value: <class 'NoneType'> None °C, len(sensor.Values): 0 
Type: Temperature, Id: 9, Name: CPU Core #1 Distance to TjMax, Value: <class 'NoneType'> None °C, len(sensor.Values): 0 
Type: Temperature, Id: 10, Name: CPU Core #2 Distance to TjMax, Value: <class 'NoneType'> None °C, len(sensor.Values): 0 
Type: Temperature, Id: 11, Name: CPU Core #3 Distance to TjMax, Value: <class 'NoneType'> None °C, len(sensor.Values): 0 
Type: Temperature, Id: 12, Name: CPU Core #4 Distance to TjMax, Value: <class 'NoneType'> None °C, len(sensor.Values): 0 
Type: Temperature, Id: 13, Name: CPU Core #5 Distance to TjMax, Value: <class 'NoneType'> None °C, len(sensor.Values): 0 
Type: Temperature, Id: 14, Name: CPU Core #6 Distance to TjMax, Value: <class 'NoneType'> None °C, len(sensor.Values): 0 
Type: Temperature, Id: 15, Name: CPU Core #7 Distance to TjMax, Value: <class 'NoneType'> None °C, len(sensor.Values): 0 
Type: Temperature, Id: 16, Name: CPU Core #8 Distance to TjMax, Value: <class 'NoneType'> None °C, len(sensor.Values): 0 
Type: Temperature, Id: 17, Name: Core Max, Value: <class 'NoneType'> None °C, len(sensor.Values): 0 
Type: Temperature, Id: 18, Name: Core Average, Value: <class 'NoneType'> None °C, len(sensor.Values): 0 
Type: Clock, Id: 1, Name: CPU Core #1, Value: <class 'NoneType'> None MHz, len(sensor.Values): 0 
Type: Clock, Id: 2, Name: CPU Core #2, Value: <class 'NoneType'> None MHz, len(sensor.Values): 0 
Type: Clock, Id: 3, Name: CPU Core #3, Value: <class 'NoneType'> None MHz, len(sensor.Values): 0 
Type: Clock, Id: 4, Name: CPU Core #4, Value: <class 'NoneType'> None MHz, len(sensor.Values): 0 
Type: Clock, Id: 5, Name: CPU Core #5, Value: <class 'NoneType'> None MHz, len(sensor.Values): 0 
Type: Clock, Id: 6, Name: CPU Core #6, Value: <class 'NoneType'> None MHz, len(sensor.Values): 0 
Type: Clock, Id: 7, Name: CPU Core #7, Value: <class 'NoneType'> None MHz, len(sensor.Values): 0 
Type: Clock, Id: 8, Name: CPU Core #8, Value: <class 'NoneType'> None MHz, len(sensor.Values): 0 
```

rmattuschka, 2021-01-23:
To use the services of the dll, your program must have adminrights.
You must run it Administrator!

PhyxionNL, 2021-01-23:
You need to run as administrator and update all hardware, including children. There's an example in the readme on how to do that.

closed 2021-01-23 by PhyxionNL

NaturalBornCamper, 2021-01-23:
AHH!! I knew it worked before! Shelved my project for too long and forgot to run as administrator, I'll add a check so it doesn't happen again, thanks friends!

Kalmat, 2022-10-24:
@NaturalBornCamper, perhaps a little late, but could you please copy-paste your code? I'm trying to use LibreHardwareMonitor in python, but I don't know how!!!

I tried to use same code as per the code to use OpenHardwareMonitor, with no luck!

```
import clr  # pythonnet, not clr module

import utils

utils.win_run_as_admin()

clr.AddReference('LibreHardwareMonitorLib')
from LibreHardwareMonitor import Hardware

handle = Hardware.Computer()
handle.Open()
print(handle, handle.Hardware)
handle.CPUEnabled = True
handle.GPUEnabled = True
handle.HDDEnabled = True
handle.FanControllerEnabled = True
handle.MainboardEnabled = True
handle.RAMEnabled = True

for i in handle.Hardware:
    i.Update()
    for sensor in i.Sensors:
        data = str(u"HW %s Type %s name %s Sensor #%i name %s: %s %s" %
                   (sensor.Hardware.HardwareType, sensor.SensorType,
                    sensor.Hardware.Name, sensor.Index, sensor.Name, sensor.Value, "\n"))
        print(data)

```