LibreHardwareMonitor/LibreHardwareMonitor #367: EVGA ICX Support and 3rd Fan

b0ssman, 2021-01-27:
The EVGA 3080 FTW has 3 fans.

Only 2 are visible (i assume those are reported by the nvidia standard access way)

there are several more sensors on the card as well as a third fan.

I unfortunately could not find any info on how to access this programmatically

the sensor data is available in e.g gpu-z

b0ssman, 2021-01-27:
[image]

dStruct, 2022-08-27:
This is also an issue for EVGA RTX 2080 Ti FTW3 Ultra Gaming (3 fan, triple slot) cards, I run Fan Control which uses Libre Hardware Monitor for it's backend, and it doesn't detect all 3 fans, or RPM signals.

I do however get 2 fans in Fan Control supposedly but the second one does nothing when I try to control it, and the first fan in Fan Control only controls 1 of the fans on the card (with matching RPM signal working).

Seems like a solid LHM issue and not on the Fan Control side.  Either way I'd love to see this fixed and I'd be happy to open a separate ticket and/or provide any additional logs or information on request.  I should mention that I do have Precision X1 installed and running for overclocking purposes, and I have tried everything aside from uninstalling it, to make sure it's not interfering with the fans, but it seems more like a LHM and/or Fan Control detection/control issue.

dStruct, 2022-08-27:
[image]

SimonZerafa, 2023-03-30:
This is also an issue with the EVGA RTX 3060 Ti FTW3 ULTRA LHR card which also has three fans.

Apparently there is an EVGA specific fan/sensor combination called "NVIDIA GeForce RTX 3060 Ti EVGA iCX" in HWInfo and described as "GPU Right Fan" and also the sensor is called "GPU Right Fan PWM"

All of the fans can be directly controlled by the EVGA Precision X1 software.

This issue also has implications for controlling the fan with FanControl by Remi Mercier as this third fan is NOT detected and cannot be controlled.

C3S4RM3W, 2023-07-22:
I wish this could be added as the EVGA RTX 3090 FTW3 has an external plug so you can plug case fans. I use the FanControl app by Remi Mercier and unfortunately there is nothing that can be done unless this support is added to librehardware

slix, 2024-02-07:
Same issue here with the EVGA RTX 3080 FTW3 Ultra.

Naratov, 2024-07-18:
Same issue with the EVGA 3070 Ti XC3 - 08G-P5-3785-KL.

superaven, 2025-05-16:
Same issue with the EVGA GeForce RTX 3090 FTW3 ULTRA HYBRID GAMING (24G-P5-3988-KR).