LibreHardwareMonitor/LibreHardwareMonitor #369: DIMM temperature probe

Cubox, 2021-01-27:
Hi,

I'm unsure if this is not already implemented, if so let me know.
I'm using a software that uses this lib, and it does not implement it yet, so if this is implemented, I'll go ask there for it.

HwInfo can read my DIMM temperature (from the DIMM temperature sensors, not all sticks have one) 
[image]

It'd be great to have this with the lib

Eno77, 2021-03-18:
I think you may close this one, since some effort is already going into this.

See:
https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/395

PhyxionNL, 2021-09-29:
Closing as duplicate.

closed 2021-09-29 by PhyxionNL