LibreHardwareMonitor/LibreHardwareMonitor #371: Network duplicates

sinancetinkaya, 2021-01-29:
Every time my pc (windows 10 x64) wakes up from hybrid sleep, network connections are getting duplicated as they can be seen in the images.
[image][image]
I have a this motherboard https://www.gigabyte.com/Motherboard/GA-AB350-Gaming-3-rev-1x/sp#sp

LiloBzH, 2021-01-31:
Same here

MSI B550 Edge Wifi

sinancetinkaya, 2021-02-07:
with the commit of 249c4c8 , network card disappears completely, instead of duplicating.

sinancetinkaya, 2021-02-07:
@sebastian-dev with the commit of 249c4c8 , network card disappears completely, instead of duplicating.

closed 2022-06-27 by PhyxionNL