LibreHardwareMonitor/LibreHardwareMonitor #374: Info (fan speeds & temps) from Asus Rampage II Gene

Epectase, 2021-02-01:
Hi, first, thanks a lot for your work !

I have tested the release 0.8.6 on my PC with a motherboard Asus Rampage II GENE (LGA1366, chipset X58 / ICH10R).

My issue is about fan speeds and optional temps.
With my current fan config, I should see 3 speeds :
> - CPU_FAN
> - CHA_FAN1
> - CHA_FAN2

OPT_FAN1 & OPT_FAN2 (see copy of motherboard documentation below) are currently connected to old fans without speed feedback so I'm not expecting something, but i would not be shocking to see these ones listed but at 0 RPM.
About "Power Fan", i'm quite surprised to see this one (and not surprised to read 0 rpm), how can it be possible to read this value as there is no communication between PSU and MB ? Maybe better to remove this one ?

For optional temps, I have plugged external sensors to the motherboard :
> - OPT_TEMP1
> - OPT_TEMP2

But they are not listed (see my print screen below). It also could be great to see Southbridge and Northbridge temperatures.

Can you add these measurements into your next release of LibreHardwareMonitor ?

Thanks in advance !

[image]
[image]
[image]

Epectase, 2021-12-31:
Hello again,

First : I'm clearly not a developer, just a random engineer who tried (and failed) to understand code.

So I have installed Visual Studio and succeeded to recompile LHM. I have also succeeded to correct the information coming from the Windbond W83667HG by modifying "SuperIOHardware.cs" to add and also to remove sensors :
<details><summary> Modified "SuperIOHardware.cs" and result once compiled
</summary>

[image]

</details>

I have understood that the information I was asking in my forst post, were not present on the W83667HG, it seems to be an other controller. So I have tried to understand and modify the parts dedicated to "embedded controllers" and seems to be an Asus trick as I can see the list of MB concerned. But after hours of different attempts, it's impossible for me, without help, to find a way to modify the embedded controller function.

I have found a very interesting topic from @zeule : https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/533

So I have discovered the DSDT register (part of ACPI), please forgive me if I'm telling bullshits. In this DSDT, I have found the data I'm looking for in the end (optional temps, additional voltages, all fans) :

<details><summary> Example of data found in the end of the DSDT
</summary>

[image]

</details>

**Can someone help me to add all these sensors to LibreHardwareMonitor ?** My goal is to see something similar as in HWInfo64  :
<details><summary> Example of HWInfo64
</summary>

[image]

</details>

Find my DSDT file (in asl and in dsl format), if it can help :
[DSDT - Asus Rampage II Gene.zip](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/files/7793933/DSDT.-.Asus.Rampage.II.Gene.zip)

Thanks a lot in advance for your help !

zeule, 2021-12-31:
Have you checked the [asus_atk0110 Linux HWMON driver](https://elixir.bootlin.com/linux/latest/source/drivers/hwmon/asus_atk0110.c)? Seems like the DSDT you shared matches it. However, the driver operates via ACPI method calls while LHM tries to access hardware directly. But anyway the driver is a better documentation source than the DSDT alone.

Epectase, 2022-01-16:
Thanks for this lib, seems to be a good base. If I understand correctly, I need to adapt the code to work in windows environment ? I will try but I think it's way over my skills...

Other question, in your post, why do you have requested some DSDT files if it was not to access them ? Not clear for me. There is no way to adapt the "Embedded controller" function you have worked for other Asus MB ?