LibreHardwareMonitor/LibreHardwareMonitor #383: Fan control only partially works on Gigabyte X570 Aorus Pro (IT8688E & IT8792E)

Snelso91, 2021-02-09:
On the Gigabyte X570 Aorus Pro, fan control only partially works on the IT8688E chip, and does not work at all on the IT8792E chip.
For fans on the IT8792E (Top Front, Front Top, Front Bottom in the images), they are initially running at the speed set by the BIOS fan curve.
[image]

However, when setting a manual fan speed to 15% on the fans, the application that I am testing with ("Fan Control") temporarily displays that the fan speed is 15% for less than a second, but shortly reverts to the BIOS fan speed of 38% afterwards. In addition, the Libre Hardware Monitor window (on the right) does not show a change at any point throughout the test implying that these fans cannot be controlled with the current version.
[image]
[image]

In addition, for the IT8688E, fans can be partially controlled which works as follows: for fans that are currently stopped (0%) on the BIOS fan curve, any speed can be set (so long as it is above the minimum speed of the fan). However, as soon as the computer heats up and the BIOS fan curve makes the fans increase their speed, they cannot be made to fan stop or decrease speed again. For example, setting a speed of 15% on the Rear and Top Rear fans has no effect if they are already spinning due to the BIOS fan curve. However, it does affect the PCH fan as that was previously at 0% before manual control was turned on.
[image]

Similarly, if 0% is set then the PCH fan will return to 0%, but the Rear and Top Rear fans will again not stop due to not being properly controlled by Libre Hardware Monitor.
[image]

However, if this is repeated when the system has cooled down and the BIOS fan curve is now in the fan stop region, then the fans will turn off.
[image]

Please could you improve support for the IT8688E and IT8792E chips so that the fans can be fully controlled?

xabelgul, 2021-02-14:
+1
Same IT8688E chip here on a Gigabyte Z390 M Gamminng-CF motherboard under Windows 10 LTSC. 
Libre Hardware Monitor  v0.8.7 recognizes and identifies it but, it has more problems, it boots badly, sometimes it takes more than 1 minute to boot or, it freezes.
Rem0o FanControl V67 does not detect the sensors and therefore does not build the  @Controls.

Thanks
[image]

jeffdamp-wave, 2021-03-02:
Just wanted to give my +1 for a fix for this.  I'm using a Gigabyte Auros Master X570 with a ITE IT8792E chip.  While I can see the fan RPMs I can't control any of them.

Xperie, 2021-03-09:
+1
Also using the Auros Master X570. Please look into this.

xabelgul, 2021-03-12:
Freezing problem at startup, solved after a clean installation.
Sorry I didn't test it before.
Version 0.8.7
Thanks

P.S. Testing now v0.8.8-ci129, looks good!

Baton34, 2021-09-28:
Same.
Z390 Aorus Pro with it8688e and it8792 chips. The fans connected to the it8792e are not controlled.

AbrilAbitia, 2021-11-26:
Same problem with IT8792E, Is there any news on this?

DocMAX, 2021-12-08:
Same here on Linux:
Gigabyte Technology Co., Ltd. B550 AORUS PRO (BIOS 14e)

it8792-isa-0 cannot be controlled (just for a second)

darkbrain-fc, 2021-12-08:
I think to know the problem. If it's true it's quite easy to understand. Simply more programs compete to control your fans, and LHM pretend to be the only one. Probably it's your BIOS that try to override LHM values. First of all you can try there to disable all Fan controls, and also verify that no others programs, like the ones shipped with you motherboard is running. If you want I've made a Fan controller service that with a simple workaround force fan control values: https://github.com/darkbrain-fc/HardwareSupervisor it's based on LHM, so you have no to warn about bios settings.

aussiehuddo, 2022-03-22:
I Have Gigabyte B550 AORUS PRO AC. IT8688E fans can be controlled but fans on IT8792E cannot. Any way to get these working?

mikehd1959, 2022-03-28:
Another Gigabyte owner, Z490 Aorus Ultra, with the IT8792E with no fan support. I'd be happy to test any updated dll for this. I have no relevant software skills to offer.

mikedmor, 2022-04-09:
Gigabyte owner with X570 AORUS EXTREME having the same issues. This seemed to be working for me previously, but recently has stopped working. Only seems to be affecting the IT8792E. Fan Control V110 is able to read the RPMs of the fans attached, but does not seem to let me manually control them at all. My IT8688E is still working correctly.

pyrlix, 2022-04-30:
Same issues on a X570 Pro. Fans on the ITE8688E seem to work just fine, but anything on the IT8792E is for now in a read-only state

BenNeumeister, 2022-05-01:
See #251

Marius8, 2022-05-01:
Are you using the Gigabyte RGB software RGB Fusion?
At least the latest version of it causes some interference in the communication with the different chips (in HWinfo i sometimes get -240°C on my ram and other nonsense values), when it is running.

mikehd1959, 2022-05-01:
> Are you using the Gigabyte RGB software RGB Fusion? At least the latest version of it causes some interference in the communication with the different chips (in HWinfo i sometimes get -240°C on my ram and other nonsense values), when it is running.

Can't speak for others, but I'm not using any other RGB software than OpenRGB.

Digital-Storm, 2022-06-18:
Busted on Gigabyte X570 xtreme, uses the same ITE IT8792E/IT8795E for the 3 fan headers. They are locked at 80%

SlimJim192, 2022-06-18:
Same issue here on B550 Aorus Pro AX rev1.0 (ITE IT8792E). 

Even gigabytes smart fan app doesnt detect the fan headers under this controller! Not that I want to use it!

PhyxionNL, 2023-03-21:
Lets continue at #IT8792E.

closed 2023-03-21 by PhyxionNL