LibreHardwareMonitor/LibreHardwareMonitor #385: Crash after sleep mode

LiloBzH, 2021-02-09:
Hi

version : 0.8.7 ci108

After each sleep mode , app is closed. No error, only windows log :

```
Application : LibreHardwareMonitor.exe
Version du Framework : v4.0.30319
Description : le processus a été arrêté en raison d'une exception non gérée.
Informations sur l'exception : System.InvalidOperationException
   à System.ThrowHelper.ThrowInvalidOperationException(System.ExceptionResource)
   à System.Collections.Generic.List`1+Enumerator[[System.__Canon, mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=xxxxxxxxxxxx]].MoveNextRare()
   à LibreHardwareMonitor.UI.TreeModel+<GetChildren>d__7.MoveNext()
   à Aga.Controls.Tree.TreeViewAdv.ReadChilds(Aga.Controls.Tree.TreeNodeAdv, Boolean)
   à Aga.Controls.Tree.TreeViewAdv._model_StructureChanged(System.Object, Aga.Controls.Tree.TreePathEventArgs)
   à LibreHardwareMonitor.UI.Node+NodeCollection.InsertItem(Int32, LibreHardwareMonitor.UI.Node)
   à LibreHardwareMonitor.UI.MainForm.SubHardwareAdded(LibreHardwareMonitor.Hardware.IHardware, LibreHardwareMonitor.UI.Node)
   à LibreHardwareMonitor.UI.MainForm.HardwareAdded(LibreHardwareMonitor.Hardware.IHardware)
   à LibreHardwareMonitor.Hardware.HardwareEventHandler.Invoke(LibreHardwareMonitor.Hardware.IHardware)
   à LibreHardwareMonitor.Hardware.Network.NetworkGroup.UpdateNetworkInterfaces(LibreHardwareMonitor.Hardware.ISettings)
   à System.Threading.ExecutionContext.RunInternal(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object, Boolean)
   à System.Threading.ExecutionContext.Run(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object, Boolean)
   à System.Threading.ExecutionContext.Run(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object)
   à System.Net.NetworkInformation.NetworkChange+AddressChangeListener.AddressChangedCallback(System.Object, Boolean)
   à System.Threading._ThreadPoolWaitOrTimerCallback.PerformWaitOrTimerCallback(System.Object, Boolean)
```

```
Nom de l’application défaillante LibreHardwareMonitor.exe, version : 0.8.7.0, horodatage : 0xef12ff45
Nom du module défaillant : KERNELBASE.dll, version : 10.0.19041.789, horodatage : 0xa9ea2bbb
Code d’exception : 0xe0434352
Décalage d’erreur : 0x000000000002d759
ID du processus défaillant : 0x502c
Heure de début de l’application défaillante : 0x01d6fe8f2be1d3ae
Chemin d’accès de l’application défaillante : D:\LibreHardwareMonitor\LibreHardwareMonitor.exe
Chemin d’accès du module défaillant: C:\Windows\System32\KERNELBASE.dll
ID de rapport : bb19115b-ade7-4f61-a0d2-f0761e266d56
Nom complet du package défaillant : 
ID de l’application relative au package défaillant : 
```
thanks !

LiloBzH, 2021-02-11:
same crash with 0.8.7 Stable

PhyxionNL, 2021-02-12:
Pinging @sebastian-dev

sebastian-dev, 2021-02-12:
i have testet this with several computers without any issues. bus this looks like an UI issue.

LiloBzH, 2021-02-12:
any a test to help  ? any log ?

LiloBzH, 2021-02-12:
It's a fairly recent problem

sebastian-dev, 2021-02-12:
Try to Enable/Disable a network interface in you system control.
Attach a Hardware Report here.

LiloBzH, 2021-02-12:
[LibreHardwareMonitor.Report.txt](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/files/5974716/LibreHardwareMonitor.Report.txt)

sinancetinkaya, 2021-02-16:
I have the same problem, either app crashes or some hardware disappears.
I'm using 0.8.7-ci105

LiloBzH, 2021-02-16:
@sebastian-dev some news ? You have time to check ? ^^ :)

sinancetinkaya, 2021-02-16:
@sebastian-dev enabling/disabling network interface doesn't cause any problem. 
The root cause is the sleep mode (I'm using the hybrid sleep)
I couldn't find any info about runtime logs, hardware report doesn't have any info about app errors.

sebastian-dev, 2021-02-17:
I have checked this with USB Devices, Bluetooth devices.. no crash. Can you disable all your network devices and then enable only one device. Try a Sleep/Resume,  do this for every device.

LiloBzH, 2021-02-17:
tests in progress. If I force the standby, I don't have a crash (maybe). I come back to you as soon as I have a result

LiloBzH, 2021-02-20:
Hello again. After a few days of testing, I think the problem is on these networks: "VMware Network Adapter VMnet1" and / or "VMware Network Adapter VMnet8"
From this morning, I test all activated networks except the 2 VMWare networks

sinancetinkaya, 2021-02-20:
I have VirtualBox installed and it has virtual network adapters on my
system 😯

LiloBzH, 2021-02-20:
@sebastian-dev 
Re,
So after tests, if i disable in windows VMware Network Adapter VMnet1 and VMware Network Adapter VMnet8 => no crash
It's ok when i disable both only, not only one

LiloBzH, 2021-03-12:
Update
Crash is coming back so it's not the only problem ... :(

tests in progress

darrencoen, 2022-04-07:
0.8.9 i have this issue as well, and i have VMWARE network devices as well.

`Source
Libre Hardware Monitor

Summary
Stopped working

Date
‎4/‎7/‎2022 7:55 AM

Status
Report sent

Description
Faulting Application Path:	C:\Users\Darren\Documents\_utilities\LibreHardwareMonitor-net452\LibreHardwareMonitor.exe

Problem signature
Problem Event Name:	CLR20r3
Problem Signature 01:	LibreHardwareMonitor.exe
Problem Signature 02:	0.8.9.0
Problem Signature 03:	c574dabb
Problem Signature 04:	mscorlib
Problem Signature 05:	4.8.4470.0
Problem Signature 06:	61b72cb6
Problem Signature 07:	33e
Problem Signature 08:	10
Problem Signature 09:	System.InvalidOperationException
OS Version:	10.0.19044.2.0.0.256.48
Locale ID:	1033
Additional Information 1:	44fa
Additional Information 2:	44fa0c85f912049a40c6e2f54e89a861
Additional Information 3:	2a92
Additional Information 4:	2a9249cd3fb286166039f82f2a1fac0d

Extra information about the problem
Bucket ID:	80fad6f22ec56020df3255b2c007a4f4 (2247953390225564916)
`

`Application: LibreHardwareMonitor.exe
Framework Version: v4.0.30319
Description: The process was terminated due to an unhandled exception.
Exception Info: System.InvalidOperationException
   at System.ThrowHelper.ThrowInvalidOperationException(System.ExceptionResource)
   at System.Collections.Generic.List`1+Enumerator[[System.__Canon, mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089]].MoveNextRare()
   at LibreHardwareMonitor.UI.TreeModel+<GetChildren>d__7.MoveNext()
   at Aga.Controls.Tree.TreeViewAdv.ReadChilds(Aga.Controls.Tree.TreeNodeAdv, Boolean)
   at Aga.Controls.Tree.TreeViewAdv._model_StructureChanged(System.Object, Aga.Controls.Tree.TreePathEventArgs)
   at LibreHardwareMonitor.UI.Node+NodeCollection.InsertItem(Int32, LibreHardwareMonitor.UI.Node)
   at LibreHardwareMonitor.UI.MainForm.SubHardwareAdded(LibreHardwareMonitor.Hardware.IHardware, LibreHardwareMonitor.UI.Node)
   at LibreHardwareMonitor.UI.MainForm.HardwareAdded(LibreHardwareMonitor.Hardware.IHardware)
   at LibreHardwareMonitor.Hardware.HardwareEventHandler.Invoke(LibreHardwareMonitor.Hardware.IHardware)
   at LibreHardwareMonitor.Hardware.Network.NetworkGroup.UpdateNetworkInterfaces(LibreHardwareMonitor.Hardware.ISettings)
   at System.Threading.ExecutionContext.RunInternal(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object, Boolean)
   at System.Threading.ExecutionContext.Run(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object, Boolean)
   at System.Threading.ExecutionContext.Run(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object)
   at System.Net.NetworkInformation.NetworkChange+AddressChangeListener.AddressChangedCallback(System.Object, Boolean)
   at System.Threading._ThreadPoolWaitOrTimerCallback.PerformWaitOrTimerCallback(System.Object, Boolean)

`

`Faulting application name: LibreHardwareMonitor.exe, version: 0.8.9.0, time stamp: 0xc574dabb
Faulting module name: KERNELBASE.dll, version: 10.0.19041.1566, time stamp: 0x0833f2d4
Exception code: 0xe0434352
Fault offset: 0x0000000000034f69
Faulting process id: 0x4fa4
Faulting application start time: 0x01d84a41425a3d42
Faulting application path: C:\Users\Darren\Documents\_utilities\LibreHardwareMonitor-net452\LibreHardwareMonitor.exe
Faulting module path: C:\Windows\System32\KERNELBASE.dll
Report Id: a69e6ceb-6028-4e2f-9647-7b7bb2c884aa
Faulting package full name: 
Faulting package-relative application ID: `

tal-sarid, 2022-10-01:
I think I found a temporary fix, in my case at least. I just removed all network-related stats from my gadget and the carshing on wakeup is gone.

noembryo, 2024-11-29:
Windows 10 22H2
Version 0.9.4
VMware Network Adapter VMnet1/8 enabled.
None of them displayed on Plots or Gadget.

Crash when coming back from sleep.. 😞