LibreHardwareMonitor/LibreHardwareMonitor #395: DIMMs temperature

lich426, 2021-02-25:
I want to implement DIMMs temperature, which category should I put it in?
fan controllers or new category?

[image]

Eno77, 2021-03-04:
I as a User would expect it in the already existing "RAM" category.
Looking forward to have my DIMM Temps shown in LHM! ;)

lich426, 2021-03-05:
[image]
I tested it with Intel cpu.
does anyone have an amd cpu?
Please help me with this test.

[LibreHardwareMonitor.zip](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/files/6088745/LibreHardwareMonitor.zip)

Eno77, 2021-03-05:
good work!

[image]

lich426, 2021-03-05:
Could you check the DIMM number with [HWiNFO (https://www.hwinfo.com)](https://www.hwinfo.com)?
thanks in advance.

Eno77, 2021-03-05:
Hmm its consistent with HWInfo, but CPU-Z shows the right Numbers:

[image]

norceno, 2021-03-05:
First of all, thank you for finally adding dimm temperatures support!
Same issue with DIMM #2 and DIMM #3 here (HWiNFO shows it this way too). MSI X570 motherboard, Ryzen 7 5800X CPU.

lich426, 2021-03-05:
smbus address of the dimm sensor is 0x18 to 0x20
your dimm address will be 0x1A and 0x1B.
therefore, index becomes 2 and 3.
you can check it through the System Infomation Viewer program. ([http://rh-software.com](http://rh-software.com))
[image]

but it doesn't matter

Eno77, 2021-03-05:
Yes, since you can rename it, it would not be an issue. Thanks again for the work!

[image]

PhyxionNL, 2021-03-05:
I just tried it and a couple of odd things. My DIMMs are named DIMM 4 and 8, which is odd as my motherboard only has 4 slots. The temperature is also a bit odd (given that room temperature is already higher):
[image]

lich426, 2021-03-05:
maybe it's because the dimm chipset is different
I don't know all of this either
I want someone to modify it after this code is merged

Pomianowski, 2021-03-17:
> does anyone have an amd cpu?
> Please help me with this test.
> 

[image]

Cubox, 2021-03-18:
[image]

Using a Ryzen 7 5800X

Cubox, 2021-03-18:
There is a DIMM8 which is not a thing and for some reasons at some point, dimm 2 was on fire.
Send me any updated file to test out if you update the code, I have been waiting for DIMM temperature in LHM for so long!

lich426, 2021-03-19:
I have already submitted the PR. (#399)

Cubox, 2021-03-19:
> I have already submitted the PR. (#399)

You can still make more commits to your repo, they will be added to the PR.
I think fixing the ghost DIMM and possibly the wrong temperature reading would be better done now to ensure the PR goes through

PhyxionNL, 2021-03-19:
I've given the PR another try and I do not think it's working correctly. My AMD sensors have the wrong number, Intel is probably similary wrong. I think that can be combined with https://docs.microsoft.com/en-us/windows/win32/cimwin32prov/win32-physicalmemory -> the Tag property gives the correct number for me (Physical Memory 1 and Physical Memory 3).

What's another concern is that, unlike the the results from @Cubox, my sensors seem to return bogus data. They're certainly not 15 and 5 degrees C. HWiNFO doesn't return anything here either, so I guess the detection isn't working well.

lich426, 2021-03-19:
smbus address of the dimm sensor is 0x18 to 0x20
therefore, the index is as follows
0x18 : 0
0x19 : 1
0x1A : 2
0x1B : 3
...

I only provided a way to get a dimm temperature sensor using a smbus.
each chipset of dimm sensors will have a different method of importing temperature sensors.
however, I don't know how to handle exceptions for chipsets.
I want someone to do elaborate coding.
I referred to this datasheet [(https://www.onsemi.com/pdf/datasheet/cat34ts02-d.pdf)](https://www.onsemi.com/pdf/datasheet/cat34ts02-d.pdf)

Eno77, 2021-03-19:
Is there a way to distinguish between DIMM with and without a temperature sensor?

lich426, 2021-03-19:
It is similar to i2cdetect of Linux.
when looking for a chipset connected to i2c, if the address and value are the same, there is a chipset connected.
ex)
[image]

the source code I worked on has SmbusDetect method.

Cubox, 2021-05-25:
Any update on this feature?

PhyxionNL, 2021-05-26:
> 
> 
> Any update on this feature?

The problems mentioned are not resolved yet.

xcasxcursex, 2021-06-04:
> does anyone have an amd cpu?
> Please help me with this test.

Working well here in FanCtrl. Missing the ability to graph it in LHM.

[image]

xcasxcursex, 2021-06-05:
> The problems mentioned are not resolved yet.

Is there any chance we could merge the existing patch which appears to work quite well in some cases, with a view to adding more sensor IC support at a later date? It would be a shame if this were held back entirely until every memory module could be supported.

Cubox, 2021-06-05:
Having used the dll from the PR since march to add DIMM capability to FanControl, I have nothing to complain about. It does see a phantom 8th stick at 13°C, but I can just ignore it.
[image]

Merging the PR would allow everyone to at least try and see if their DIMMs are properly picked up, and use those values until this is entirely fixed.

PhyxionNL, 2021-06-22:
@Cubox Can you try this one: https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/actions/runs/960007578
I don't seem to have any DIMM with temperature sensor here.

xcasxcursex, 2021-06-22:
> @Cubox Can you try this one: https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/actions/runs/960007578
> I don't seem to have any DIMM with temperature sensor here.

I'm afraid this one broke it for me. No temperatures in this version. All of them have been good for me so far otherwise.

Eno77, 2021-06-22:
Broken for me too, no DIMM Temps with this one.

Cubox, 2021-06-23:
> 
> 
> @Cubox Can you try this one: https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/actions/runs/960007578
> I don't seem to have any DIMM with temperature sensor here.

not a single DIMM :(

PhyxionNL, 2021-06-23:
@xcasxcursex @Eno77 @Cubox Anything with this one?
https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/actions/runs/964965830

Cubox, 2021-06-23:
[image]

PhyxionNL, 2021-06-23:
That's more like it (I hope it also works for the others). It isn't detecting my DIMMs with this anymore either. There's still some major cleaning required for the PR and temperature resolution should probably also be added.

Eno77, 2021-06-23:
[image]

Looks OK,
Enumeration like mentioned in the very first build is a bit strange (since my DIMMS should be 1 & 3 (or 2 & 4 if not 0 based) , but I could live with it.

PhyxionNL, 2021-06-23:
> Looks OK,
> Enumeration like mentioned in the very first build is a bit strange (since my DIMMS should be 1 & 3 (or 2 & 4 if not 0 based) , but I could live with it.

This can probably be read from SMBIOS. For example at my PC (no temperature sensors):

DIMM 0, P0 CHANNEL A - no module
DIMM 1, P0 CHANNEL A - module found
DIMM 0, P0 CHANNEL B - no module
DIMM 1, P0 CHANNEL B - module found

This would also fix my weird naming (currently detects 4 and 8 instead of 2 and 4). The SMBIOS can also be used to detect AMD or Intel to avoid WMI completely.

Eno77, 2021-06-23:
OK, HWMonitor has the same "issue" - so I would not mind, if there is no "fix".

I could have a look and play around a bit. if you point me to the right branch and source files ;)

PhyxionNL, 2021-06-23:
Here: https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/pull/468

Eno77, 2021-06-24:
OK, from the SMBios one could even show the actual Manufacturer (instead of "GenericMemory")

[image]

But I surrender to the complexity / time to work into SMBios readouts.
Using the Address from WMI as the DIMM Number would be OK for me ;)

xcasxcursex, 2021-06-24:
Honestly, it would be great to see the correct numbers on the DIMMs, but honestly, I don't really care. I've never seen any other software get it right; accordingly I don't think that should block this change; and if I need to tell one DIMM from the other I'm going to be in there replacing hardware anyway.

PhyxionNL, 2021-06-24:
> Honestly, it would be great to see the correct numbers on the DIMMs, but honestly, I don't really care. I've never seen any other software get it right; accordingly I don't think that should block [...]

This data already exists in SMBIOS from LHM so it would not be too much work to base it on that 🙂

xcasxcursex, 2021-06-24:
Sorry, just noticed something missing: The ability to disable this: 
[image]

Eno77, 2021-06-24:
Found it.
But dont see, how the MemoryDeviceList is acessible from within the "GenericMemory" implementation.

VitaminCpp, 2023-12-14:
Still doesn't work correctly, but HWINFO is able to display temperatures correctly:
[image]
[image]

DearTanker, 2024-04-11:
I am very disappointed that the memory temperature cannot be identified yet. The memory temperature of DDR5 has become one of the serious problems that must be solved in the new era of computer cooling.

PhyxionNL, 2024-04-11:
> I am very disappointed that the memory temperature cannot be identified yet. The memory temperature of DDR5 has become one of the serious problems that must be solved in the new era of computer cooling.

Feel free to submit a PR if you're so disappointed.

Cubox, 2024-04-11:
> I am very disappointed that the memory temperature cannot be identified yet. The memory temperature of DDR5 has become one of the serious problems that must be solved in the new era of computer cooling.

While I am also sad I must keep using an old build of LHM to get DIMM temperature readout, the implementation of DIMM temperature reading isn't as simple as it seems to be, and the demand is small.

Even with me willing to set a bounty for this issue, the amount I'm willing to pledge is quite low due to the sufficient solution I have today.

As @PhyxionNL said, this being open source, feel free to implement it yourself and submit your code. I still use @lich426's code today for my own build.

DearTanker, 2024-04-12:
> > I am very disappointed that the memory temperature cannot be identified yet. The memory temperature of DDR5 has become one of the serious problems that must be solved in the new era of computer cooling.
> 
> Feel free to submit a PR if you're so disappointed.

I don't know how to code, I'm just expressing my personal feelings. Sorry if it causes any inconvenience.

ouyangyiluo, 2024-05-19:
This works for me

[image]

https://www.renesas.com/us/en/products/memory-logic/memory-interface-products/ddr5-solutions/spd5118-spd-hub-ddr5-memory-modules
https://www.alldatasheet.com/datasheet-pdf/pdf/1574989/RENESAS/SPD5118.html
[image]

[image]

ouyangyiluo, 2024-11-04:
AMD+DDR5
This works on my device
[image]
[image]
https://github.com/ouyangyiluo/LibreHardwareMonitor/tree/amd_am5_ddr5_dimm

mahpooya, 2024-12-10:
@ouyangyiluo Can you please provide me a compiled release of ryzen ddr5 branch 330

mahpooya, 2024-12-19:
> > @ouyangyiluo Can you please provide me a compiled release of ryzen ddr5 branch 330
> 
> The code in this branch is not perfect. It only supports AMD, uses a fixed AMD SMBus address, and has only been tested to a limited extent on my device. You should be aware that running this program may pose a risk of hardware issues.
> 
> [publish.zip](https://github.com/user-attachments/files/18146644/publish.zip)

Worked for me like a charm in FanControl

Although sometimes zero and negative values are announced for just a moment, I solved the problem by taking the average of the last 3 seconds

And now I don't need HWInfo Plugin and running in background and causing some negative effects on performance.
Thanks a lot. Hope someday it will be pulled and merged in main project

CyberRSR, 2024-12-22:
@ouyangyiluo 
> I made some changes that should fix the problem with zero values. But for the problem with negative values, can't figure out what is causing it.

Need to merge to LHM and FanControl

killo3967, 2025-02-27:
I have an Asus Prime B760-Plus D4 with a Nuovoton NCT6798D-R with a Corei5-13400T and Corsair DDR4-3600 (Micron) and for me the memory temperatura don't display with the last compilation. 
But with the  the version you post [here](url) the program show's the rempoeerature from my tho banks of dimm and a third one that i don't know what is.
That is the version that i'm running now.  
Is there a forecast of when it will be implemented in the final version?
Whatever you need from me, whatever testing you want to do, I am willing to do it.

yw4z, 2025-03-28:
@ouyangyiluo do you mind sending changes as draft commit? maybe other ones review code and find a solution. i tested that and worked on g-skill 6400(24x2) kit

spec-chum, 2025-04-18:
> > > > [@ouyangyiluo](https://github.com/ouyangyiluo) Can you please provide me a compiled release of ryzen ddr5 branch 330
> > > 
> > > 
> > > The code in this branch is not perfect. It only supports AMD, uses a fixed AMD SMBus address, and has only been tested to a limited extent on my device. You should be aware that running this program may pose a risk of hardware issues.
> > > [publish.zip](https://github.com/user-attachments/files/18146644/publish.zip)
> > 
> > 
> > Worked for me like a charm in FanControl
> > Although sometimes zero and negative values are announced for just a moment, I solved the problem by taking the average of the last 3 seconds
> > And now I don't need HWInfo Plugin and running in background and causing some negative effects on performance. Thanks a lot. Hope someday it will be pulled and merged in main project
> 
> I made some changes that should fix the problem with zero values. But for the problem with negative values, can't figure out what is causing it. [publish.zip](https://github.com/user-attachments/files/18204776/publish.zip)

Can confirm this works fine on latest Fan Control as well, I've not seen it go negative or 0 yet - ASUS B650E-E with 2 x 32GB Corsair DDR5.

Going negative is, I'm assuming, a sensor read error above `int.MaxValue`, so shows negative?

yw4z, 2025-04-22:
@ouyangyiluo Can you your update library with latest approved pull request for MSI X870 ? Thanks

spec-chum, 2025-04-24:
@ouyangyiluo  I can reproduce the 0C bug every time now.

Loading OpenRGB puts the DIMM temps to 0 when it scans devices, every time, however it's corrected by loading anything that probes the sensor again, I've tried OCCT, HWiNFO and AIDA64 and they all bring it back to life and it stays on until I load OpenRGB again (was fine for 2 days and PC has been on 24/7).

Hope this helps a little in tracking down why.

mahpooya, 2025-04-28:
> Let me know if someone could assist with testing and I will get a version ready to test. (This may take a day or two to verify all is still working as expected)

I am on X670E HERO and 7800x3d and 4090
Please provide a compiled release like ouyangyiluo does above and i can test on my system

Eno77, 2025-04-28:
I got an exception - unfortunately its obfuscated, but perhaps this helps:

`[28.04.2025 16:20:40.294] [DEBUG] Entering DoTest.
[28.04.2025 16:20:40.347] [DEBUG] ***** Driver is open. *****
[28.04.2025 16:20:40.347] [DEBUG] Detecting SMBuses - available detection methods are 4.
[28.04.2025 16:20:41.553] [DEBUG] Info:
(0x80004003) Object reference not set to an instance of an object.

Details:
   at a.D.A()
   at A.a.a()
[28.04.2025 16:20:41.554] [DEBUG] ***** Closed driver. *****`

Blacktempel, 2025-04-28:
@Eno77 that is an NVAPI issue with NVIDIA SMBus. I will check that after dinner or just outcomment it as it is not necessary for the tests.
Which GPU do you have ?

Eno77, 2025-04-28:
Thanks for the quick fix, it works now. (Yes, I got an AMD 6900XT GPU - nothing from NVIDIA)

Temperatures from my modules look reasonable:

```
[28.04.2025 19:33:29.276] [DEBUG] Temperature for Module at address 0x52 is: 40,25°C / 104,45°F.
[28.04.2025 19:33:29.539] [DEBUG] Temperature for Module at address 0x53 is: 41,75°C / 107,15°F.
```

The whole log file is attached, let me know if you need additional info.

[Log.zip](https://github.com/user-attachments/files/19945305/Log.zip)

Blacktempel, 2025-04-28:
That's great. I can see both DDR4 are in the last two slots and seem to be from G.Skill. It's not reading the model at the moment, so I only see the manufacturer.
I don't need more information right now, just more testers + Log files to verify it works on more systems than I already tested :)
You can compare the values to (for example) "SIV - System Information Viewer" (When started: arrow next to "Sensors" and "DIMM Status").

For now it looks good. I will check how I can integrate it properly.

[25 more replies]