LibreHardwareMonitor/LibreHardwareMonitor #396: SATA HDD Temp over SAS adapter

alex6dj, 2021-02-25:
Hi I´am using another app called [FanControl](https://github.com/Rem0o/FanControl.Releases/) and it is using in the background this library to control several fan according to several hardware temperature sensors.

Actually it lacks support for SATA HDD temperature sensing over SAS because this library don´t provide that support actually. [No sensing HDD temp when connected to SAS controller](https://github.com/Rem0o/FanControl.Releases/issues/109).

Is any future plan to support this configuration.

alex6dj, 2021-04-14:
After some test the issue is not about HDD over SAS controller, but whe hard drivers are added to Microsoft Storage Spaces it can't be checked with this library. I don't know if in the background WMI is used, the this should be the problem. The only way I can found to get the drive info is using PowerShell.

`Get-PhysicalDisk | Get-StorageReliabilityCounter | fl`

sun1ight, 2021-05-20:
Same problem, all disks added (AHCI, NVME) in Storage Spaces it can't checked with this library.

alex6dj, 2022-04-18:
For now it seems that the only solution is to get the needed info using PowerShell.

For my use case I develop a simple plugin to be used with [FanControl](https://github.com/Rem0o/FanControl.Releases/) app. The plugin get the list of drives used in storage spaces and present the temperature values to the app .

Plugin: [FanControl.StorageSpacePlugin](https://github.com/QAppsSoft/FanControl.StorageSpacePlugin)

SoftwareMagicIT, 2022-08-07:
Good, but OpenHardwareMonitor latest show temperature of HDD and M.2 Sata in my pc

> For now it seems that the only solution is to get the needed info using PowerShell.
>
> For my use case I develop a simple plugin to be used with [...]

alex6dj, 2022-12-17:
Reopening because the problem was parcially resolved. The library show the HDD temperature when connected to the motherboard SATA connectors, but not when connected to an SAS card adapter.

alex6dj, 2022-12-17:
And last LibreHardwareMonitor version dont show temperature info related to physical discs in Microsoft Storage Spaces. But shows some info for virtual disks.

fbcl00, 2023-09-17:
> Reopening because the problem was parcially resolved. The library show the HDD temperature when connected to the motherboard SATA connectors, but not when connected to an SAS card adapter.

Got the same issue. Anyone got a fix?

Landcross, 2024-12-27:
Is there any update on this? I also would like to get my HDD temperatures via LibreHardwareMonitor into FanControl, but the temps don't show up in LibreHardwareMonitor. I also have my (Sata) HDDs connected via a SAS adapter. However, I am _not_ using Microsoft Storage Spaces. Using other software, the S.M.A.R.T. data shows the temperature just fine.

closed 2025-12-30 by PhyxionNL