LibreHardwareMonitor/LibreHardwareMonitor #401: Add support for external coolant temperature sensor

ahodges-git, 2021-03-09:
I have a coolant temperature sensor that plugs into a 2 pin header (T_Sensor) on my Asus Crosshair VIII Impact motherboard.

I'm keen to see if this can be supported so I can make use of it in https://github.com/Rem0o/FanControl.Releases

Motherboard manual:
[image]

Appears in HWiNFO64:
[image]

stevepositiv, 2021-03-11:
I have the same issue on my ROG X570-I.

@contributors: Is there any way I could help to get this sensor readable?

Ja50n-Kru, 2021-03-12:
As seen in HWiNFO 64, that sensor is on the ASUS EC chip - see issue #282

PhyxionNL, 2021-03-18:
> 
> 
> As seen in HWiNFO 64, that sensor is on the ASUS EC chip - see issue #282

Closing this in favor of that.

closed 2021-03-18 by PhyxionNL