LibreHardwareMonitor/LibreHardwareMonitor #402: CPU load values is incorrect...

Kliodna, 2021-03-11:
Hello. To begin with, thanks for your work and such useful software.
I am using your library to create my own little home project and ran into a very frustrating situation.
The processor core load indicators display indicators that are completely untrue. I tested this on several computers with processors of different Intel generations and the result turned out to be the same every time.
Thanks in advance for your help and a possible answer

`

            computer.Open();
            computer.Accept(new UpdateVisitor());

            info = string.Empty;
            foreach (IHardware hardware in computer.Hardware)
            {

                info += string.Format(" Hardware: {0} ({1}) \r\n", hardware.Name, hardware.HardwareType.ToString().ToUpper());

                foreach (IHardware subhardware in hardware.SubHardware)
                {
                    info += string.Format("\tSubhardware: {0} ({1}) \r\n", subhardware.Name, subhardware.HardwareType.ToString().ToUpper());

                    foreach (ISensor sensor in subhardware.Sensors)
                    {
                        info += string.Format("\t\tSensor: {0} ({1} - {2}) = {3} \r\n", sensor.Name, sensor.SensorType, sensor.Index, sensor.Value);
                    }
                }

                foreach (ISensor sensor in hardware.Sensors)
                {
                    info += string.Format("\tSensor: {0} ({1} - {2}) = {3} \r\n", sensor.Name, sensor.SensorType, sensor.Index, sensor.Value);
                }
            }

            computer.Close();

`

[image]

CodeZ1LLa, 2022-01-31:
I ran into the same issue. 
The problem is, that LHM shows the load of both, logical and physical cores. 
This means, if multi-threading is ON and you have 4 physical cores, LHM will count as there are 8 cores. 
Now, when you put some kind of load on the core, the Windows Scheduler will prefer a logical core, which means, 1 of 2 threads will idle under most conditions. 

The solution is to poll every 2nd thread. Say, if you want to know the load of core 1, search for ("CPU Core#2"), for core 2, search for ("CPU Core#4"), and so on.

CodeZ1LLa, 2022-01-31:
Also, I would recommend you to use my optimized SensorService class. It's much more readable than your code, and you can easily add/delete what you need 
https://github.com/CodeZ1LLa/LibreHardwareMonitor-Optimized-Sensors-Scanner/blob/main/SensorService.cs

closed 2022-09-23 by PhyxionNL