LibreHardwareMonitor/LibreHardwareMonitor #415: This project is dead .-.

LaTecnologia24, 2021-03-31:
This project is dead .-.

LaTecnologia24, 2021-03-31:
Reported 2 important issue no answer or fix

https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/410

DjordjeMandic, 2021-04-01:
Read the [License](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/blob/master/LICENSE)
```
************************************************************************
*                                                                      *
*  6. Disclaimer of Warranty                                           *
*  -------------------------                                           *
*                                                                      *
*  Covered Software is provided under this License on an "as is"       *
*  basis, without warranty of any kind, either expressed, implied, or  *
*  statutory, including, without limitation, warranties that the       *
*  Covered Software is free of defects, merchantable, fit for a        *
*  particular purpose or non-infringing. The entire risk as to the     *
*  quality and performance of the Covered Software is with You.        *
*  Should any Covered Software prove defective in any respect, You     *
*  (not any Contributor) assume the cost of any necessary servicing,   *
*  repair, or correction. This disclaimer of warranty constitutes an   *
*  essential part of this License. No use of any Covered Software is   *
*  authorized under this License except under this disclaimer.         *
*                                                                      *
************************************************************************

************************************************************************
*                                                                      *
*  7. Limitation of Liability                                          *
*  --------------------------                                          *
*                                                                      *
*  Under no circumstances and under no legal theory, whether tort      *
*  (including negligence), contract, or otherwise, shall any           *
*  Contributor, or anyone who distributes Covered Software as          *
*  permitted above, be liable to You for any direct, indirect,         *
*  special, incidental, or consequential damages of any character      *
*  including, without limitation, damages for lost profits, loss of    *
*  goodwill, work stoppage, computer failure or malfunction, or any    *
*  and all other commercial damages or losses, even if such party      *
*  shall have been informed of the possibility of such damages. This   *
*  limitation of liability shall not apply to liability for death or   *
*  personal injury resulting from such party's negligence to the       *
*  extent applicable law prohibits such limitation. Some               *
*  jurisdictions do not allow the exclusion or limitation of           *
*  incidental or consequential damages, so this exclusion and          *
*  limitation may not apply to You.                                    *
*                                                                      *
************************************************************************
```

If someone has time and will and knows how to fix they will probably contribute to the project, the owner of the project cannot do everything alone. You can bump the issues if you wish for people to notice or do a lot of search on web and look into source code of the project and try to fix it yourself then create a pull request. I also could say that project is dead because there was no authentication support for web interface. Try to do it by yourself and create pull request / start a discussion :)