LibreHardwareMonitor/LibreHardwareMonitor #422: ASUS ROG Strix B450-I - Fan Control v74 does not detect AIO pump (PWM)

Dutchy735, 2021-04-09:
Title says it all.

i recently started using this software and i'm really liking it. this however is a major disappointment. The AIO does pop up in the BIOS and even displays its RPM. I believe it has something to do with the hardware libraries. However, I'm a big noobie when it comes to these things. I tried everything that I could do myself though. I'm running the latest BIOS version and installed al windows updates.

Hardware:

Asus ROG Strix B450-I Gaming (mini ITX)
AMD Ryzen 7 3800XT
Custom Loop AIO pump on PWM mode, on the dedicated AIO PWM Header on the motherboard

Software:

Windows 10 Pro
Fan Control V73

If anyone could help that would be magnificent!

[image]

KaktusKontrafaktus, 2022-04-22:
I have a similar issue on an ASUS B450-F Gaming.

The Board has 6 PWM headers:
CPU
CPU_OPT (no separate PWM control, mirrors the PWM setting of the CPU header)
Chassis 1-3
Pump

LHW (0.8.9) detects 5 headers, 4 of which (CPU & Chassis 1-3) work fine (PWM control & and same RPM value as HWiNFO64), 
the 5th always reports 0 RPM and the fan control slider does nothing.

Software:
Windows 10 Pro
LHW 0.8.9
Fan Control V110

Svek, 2022-06-14:
It would be helpful to show where the fan sensor values are showing in HWiNFO. As in, is it listed under the board, or under the ASUS EC? If it is under the ASUS EC, it's because the board is not explicitly identified in the LibreHardwareMonitor code to take values from the ASUS EC.

Just speculation, but if you could confirm it, it's a rather simple update to the code to support this motherboard.

KaktusKontrafaktus, 2022-06-14:
> It would be helpful to show where the fan sensor values are showing in HWiNFO. As in, is it listed under the board, or under the ASUS EC? If it is under the ASUS EC, it's because the board is not [...]

[image]

Fan #1-4 & Fan Control #1-4 work fine. 
Fan/Control #5 should be the pump header, no rpm reading and the set pwm value has no effect.
There should be a fan #6 without a corresponding separate control, the "CPU_OPT" header mirrors the pwm setting of "CPU" (Fan #1).