LibreHardwareMonitor/LibreHardwareMonitor #429: ASUS ROG STRIX X570-I GAMING - Chipset Fan

adamhl8, 2021-04-17:
It doesn't seem like there is support for monitoring the Embedded Controller on ASUS boards (or at least on mine). HWiNFO64 reports the chipset fan speed under "ASUS EC".
[image]
Ultimately I'd like to control that fan's speed with [FanControl](https://github.com/Rem0o/FanControl.Releases).

zeule, 2021-09-21:
There is EC support now in the LHM, and there is a configuration for [ROG_STRIX_X570_E_GAMING](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/blob/095cce039462028b2602e04fa05ac7c54030d430/LibreHardwareMonitorLib/Hardware/Motherboard/Lpc/EC/EmbeddedController.cs#L56). Maybe you can check it out and add ROG STRIX X570-I GAMING?

shewless, 2022-06-26:
@zeule is looks like someone already added support to read the chipset fan speed for the ROG STRIX X570-I GAMING.  I am working on adding the Chipset and VRM temperature as well.  
Does LibreHardwareMonitor have support to "Control" things on Asus EC or just "Read" sensors?  To me it looks like just reading for now.

zeule, 2022-06-26:
@shewless  I'm not aware of any way to control fans via EC. Does ASUS expose control for that fan in BIOS?

shewless, 2022-06-26:
@zeule. It isn't directly exposed in the BIOS of the strix x570i but perhaps on some models it is.  You can actually change the fan curve by following these instructions: https://www.reddit.com/r/Amd/comments/nu59wl/modifying_pch_fan_curve_on_nonmodded_asus_x570/
Modifying a UEFI variable:
setup_var_cv QFan 0x3a 0x01 0x50

jkroepke, 2023-05-14:
Hi,

I have a ROG STRIX B550-I GAMING and the same question. Through an BIOS mod, its possible to control the Chipset fan on a PWM base with Curves.

In general its possible to control the fan.