LibreHardwareMonitor/LibreHardwareMonitor #433: Fan Control: some sensors disappear unless I open CPUID HWMonitor

TiffanySerenity, 2021-04-27:
Hi,

I've used Fan Control since v75 (currently at v77) and I have an issue where some sensors disappear after going into sleep mode, or if I have restarted Windows and open the Fan Control GUI. I added the FanControl.HWinfo plugin and ran HWinfo v7.0.2 and enabled  Driver Shared Memory & installed the driver with permanent option (SensorsSM=1, PersistentDriver=1)
I thought this may help with hardware sensor support but the only trick that allows me to get my sensors back is to run CPUID's HWMonitor (currently v1.44) and then all sensors appear after scan and all is well.

System Info:
OS: Windows 10 Home (up to date as of April 23, 2021)
Mother Board: ASUS PRIME A320M-C R2.0 AMD Ryzen AM4
CPU: AMD Ryzen 5 3600 (6 core / 12 thread) Unlocked
Corsair Vengeance LPX 32GB (2x16GB) DDR4 2133

Software:
Fan Control v77 + FanControl.HWinfo plugin
HWinfo v7.0.2 (settings - SensorsSM=1, PersistentDriver=1)
hwmonitor v1.44 (Released April 27, 2021)

I am wondering why HWMonitor makes the missing sensors appear, and I have tried it with and without the FanControl.HWinfo plugin, and I am at a loss for why this happens.

If there are any other settings for HWinfo plugin or hwinfo software besides sensor shared memory and loading the driver and checking the plugins folder that the hwinfo plugin is loaded, are there other hardware aware plugins I should try?

Thanks so much for any help - and I love the Fan Control software's power and flexibility, but am hoping to address this issue.

--

realsmoky, 2021-05-10:
Hey, T! 

I am currently also having problems with one fan header not being recognized at all, i.e. the control for it is not there, but I can see the fan and its RPM (and I can also see a fan that I don't have and is always at 0 (???) but whatevs).

Can you explain your workaround in more detail, i.e. exactly which file I need to paste where? I am completely new to Github and have 0 coding experience, the most I did is change INI settings to make some games run better, so i would really appreciate a detailed explanation for the workaround. I kinda tried to do it already, but either I did it wrong, or your workaround doesn't work for me.

Thanks!

TiffanySerenity, 2021-05-11:
Once I have rebooted, before opening the fanControl GUI, I launch CUPID's HWMonitor program, minimize it, then open FanControl & all of my sensors are there. However, if the sensor has disappeared after computer goes into sleep mode, then I first close, and re-open HWMonitor, then check fancontrol to see if my sensors are there, if I just leave the previously opened HWMonitor alone after sleep without re-opening it, then I sometimes get the disappearing sensor. Hope that helps, Like I said, its a convoluted way to work around the issue, and I'm hoping to find a better way using hwinfo's plugin / driver support so I don't have to open HWMonitor every time prior to opening FanControl.  Hope this helps! :)

Svek, 2022-06-17:
A quick update... I've added some code to get into this issue further.
Some triage and a bit more testing might be needed. (https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/pull/758)

Svek, 2022-07-04:
Would you mind testing the latest release. It slightly modified the usage of the mutex permissions.

I believe that we should add some logging into the code for this section so that there is something in the logs, but that might be something further down the line if the latest patch is not sufficient.