LibreHardwareMonitor/LibreHardwareMonitor #436: Gigabyte Aorus X570 Pro (not wifi) Rev 1.0 Case Fan Headers Fan6_PUMP & FAN5_PUMP bios settings aren't getting overridden

trif55, 2021-05-07:
I reset the BIOS to default having set these two fan headers to be controlled from system temp instead of CPU temp (Being able to be used as auxiliary pump headers they have this option) anyway, now they're back to CPU temp they're wildly up and down again, the software can read the fan speed and the power % but can't effect the power percentage when I change it with a curve or to 0% or 100% is there any fix for this? thanks,

the Gigabyte SIV software controls them but I'd like to have them linked to GPU temps

Thanks

ccodykid, 2022-04-22:
Same issue on Gigabyte B550 Aorus Pro V2.

MazzicCron, 2022-04-23:
I am running the Aorus X570 and seeing some weird issues as well...I'm using Fan Control, which essentially puts a UI over LHM, and the System Fan 4, System Fan 5 Pump, and System Fan 6 Pump all are dodgy. They either don't show at all or they do show, but I cannot interact with them at all.

Dirkxke, 2022-05-08:
I have the X570 Aorus Pro to and I am not able to control "System 4, System 5 Pump, System 6 Pump".
All the other fans are controllable.

The controller of fan 4,5 and 6 is a ITE IT8792E.
[image]