LibreHardwareMonitor/LibreHardwareMonitor #441: FanControl recognizing only 2/3 mainboard headers

realsmoky, 2021-05-10:
Hello, everybody!

I am a huge _fan_ of FanControl (no pun intended). However, I am having a slight problem.

Right now, FanControl  is only displaying 2 out of 3 fan headers on my motherboard (not counting the CPU header).

In FanControl, I see a total of 4 fans I can control: the GPU fan, the CPU fan, and 2 motherboard fan headers, but the third one is missing.

I have a Gigabyte GA-B150M-D3H motherboard. https://www.gigabyte.com/Motherboard/GA-B150M-D3H-rev-10#ov

In the bios, I can configure the third header, however, I am unable to find the third header with your software.

I will attach a screen shot of how it looks on my end. You will see that I am not using the GPU fan to cool the GPU, I am instead using a normal case fan strapped to it and plugged in into a motherboard header.

Can you help me out with this?

Thanks in advance!

[image]
[image]

sp991, 2022-10-20:
Hi there,

I have the exact same issue with only 2 fan header out of 3 on the motherboard showing + CPU fan & OPT and GPU. The missing fan header is the one called System Fan 3 in the application provided by Gigabyte and it's called System 3 in HWiNFO64. See bellow screenshot showing Fan Control vs HWiNFO.
My motherboard is the Gigabyte z170mx gaming 5 [https://www.gigabyte.com/Motherboard/GA-Z170MX-Gaming-5-rev-10#ov](url)

I believe this would address the issue on many different gigabyte motherboard.
[image]

sp991, 2022-10-25:
Hello, I've manage to find how to solve the exact same issue for my motherboard (GA-Z170MX Gaming 5) and if you can give me the code name of you motherboard as it appears in LibreHardwareMonitor or HWiNFO I could take a look at the code to solve yours too.
[image]
[image]