LibreHardwareMonitor/LibreHardwareMonitor #446: Add support for Lenovo 3717 - Nuvoton NCT6686D

soxhi8, 2021-05-14:
So I am trying to add support for this Lenovo pre-built mainboard:
Intel Comet Lake B460 WIN DPK, Part No: 5B20W27646
https://download.lenovo.com/Images/Parts/5B20W27646/5B20W27646_A.jpg

Another tool is able to read it fine (Fan2, aka AUX2_FAN missing due to no fan being attached):
[image]

I was trying to get it working by replicating the changes from this pull request, but to no avail:
https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/pull/406/files

Would someone be willing and able to assist me with the necessary changes or implement them?

soxhi8, 2021-05-20:
Log-file for mainboard and Nuvoton chip:
`
SMBios Version: 3.2

BIOS Vendor: LENOVO
BIOS Version: O4NKT19A
BIOS Date: 16.12.2020
BIOS Size: 16 MB

System Manufacturer: LENOVO
System Name: 90NC00LBGE
System Version: Legion T5 28IMB05
System Wakeup: PowerSwitch

Motherboard Manufacturer: LENOVO
Motherboard Name: 3717
Motherboard Version: SDK0R32862 WIN 3258211058107

LpcIO

Chip ID: Unknown Winbond / Nuvoton / Fintek with ID 0xD441 at 0x2E/0x2F

Chip ID: Unknown ITE with ID 0xD441 at 0x2E/0x2F

Chip ID: Unknown SMSC with ID 0xD441 at 0x2E/0x2F`

PhyxionNL, 2021-05-22:
Did you change the value of NCT6686D too? NCT6686D should be 0xD440. You'll also need to change LpcIO to the correct values. For the rest it should follow NCT6687D and NCT6683D going by the sources I've found.

Sevaeron, 2022-10-27:
Were you able to figure this out? just received my Z690 Phantom Gaming-ITX/TB4 and it uses the same chip, and shows no sensors.