LibreHardwareMonitor/LibreHardwareMonitor #450: Unmanaged memory leak?

nangusa, 2021-05-21:
Hi,

I'm using LibreHardwareMonitorLib to develop a Windows service. I noticed an steady increase in memory usage over time, which is never reclaimed.

Profiling memory usage, I can see the memory increase belongs to the unmanaged code, but I'm unable to pinpoint it.

Running the Libre Hardware monitor app it's the same, so I guess it's the Lib and not my service. I've set the internal values collection to zero, and it correlates what I'm seeing when profiling memory, so I think that's not the memory leak problem I'm seeing.

Anyone noticed something similar?

Thanks.

Pomianowski, 2021-05-23:
## Case study
Hi, you're right, in fact, LibreHardwareMonitor slowly takes up more and more memory over time.

[image]

My NET5 LibreHardwareMonitorLib-based application exhibits a similar property. However, the number of Objects is also increasing in it, caused by constantly calling BindingExpression for a TextBlock.

[image]

## Why?
One possible source of the problem is that the Sensor stores all of its changes. In this way, `Sensor.Values` continues to grow.

## Possible solution
Implement the `StoreValues = false` attribute
or
`Sensor.Clear()` that removes all values from the Enumerable.

nangusa, 2021-05-23:
Hi, thanks for commenting.

The Sensor.Values Enumerable stores values in a time span. Default timespan is 1 day, so that's why you are seeing the increment in objects. You can control that timespan by setting the TimesValueWindow, and if set to TimeSpan.Zero, the values enumerable is cleared, so you can reclaim that memory back.

In my opinion, this steady increase in objects and memory is expected and it is not the problem I'm trying to describe. If I set the TimesValueWindow property to Timespan.Zero, the Sensor.Values Enumerable don't increase and I don't see an increase in objects, neither memory usage, in the debugger.

What I'm reporting here, and I'm trying to determine, is a steady increase in the unmanaged memory allocation which increase over time and is never reclaimed by the system. I suspect there are some objects not cleared somewhere in the library code, may be the low level hardware access functions used to read the different super IO chips, but I'm unable to find the culpruit yet.

 Thank you.

kolos, 2021-05-24:
Maybe the NVIDIA driver?

As seen here: https://www.hwinfo.com/forum/threads/memory-leak-hwinfo-using-10g-memory.6937/

nangusa, 2021-05-24:
> Maybe the NVIDIA driver?
> 
> As seen here: https://www.hwinfo.com/forum/threads/memory-leak-hwinfo-using-10g-memory.6937/

Hi, thanks for the link. I will try to see if the problem is the call to that NVidia function and report back.

Thank you again for the insight.

nangusa, 2021-05-24:
> Maybe the NVIDIA driver?
> 
> As seen here: https://www.hwinfo.com/forum/threads/memory-leak-hwinfo-using-10g-memory.6937/

It is, indeed, the memory leak I was reporting. Thank you for pointing it out.

The memory leak is a bug in the NVidia driver when accesing the NvidiaML functions to get power metrics. It's present in the R460 driver branch apparently, and solved in R470.

Instead trying several driver versions, I compiled LibreHardwareMonitorLib to not use those NvidiaML functions and the memory leak is gone. The downside is you don't get power and throughput usage for the video card, but it's a workaround until NVidia solves the problem, or if you need to use a bugged driver version for whatever reason.

TLDR: Memory leak is not in LibreHardwareMonitor itself, but in the Nvidia R460 branch driver. This issue can be set as closed. Thanks.

closed 2021-05-24 by nangusa