LibreHardwareMonitor/LibreHardwareMonitor #460: Crashing when computer wakes up

Zlikster, 2021-06-03:
App crashes when computer wakes up from sleeo (Ryzen 5800x/GB Aorus x570 w10 2002)

PhyxionNL, 2021-06-03:
Can you post the error message from Windows' Event Viewer?
It should be in Windows Logs > Application and then filter by Error. It's source is likely .NET Runtime.

Zlikster, 2021-06-03:
sure!

Faulting application name: LibreHardwareMonitor.exe, version: 0.8.7.0, time stamp: 0xf75f5c59
Faulting module name: KERNELBASE.dll, version: 10.0.19041.964, time stamp: 0x812662a7
Exception code: 0xe0434352
Fault offset: 0x0000000000034b89
Faulting process id: 0xb54
Faulting application start time: 0x01d7588dcd2a256f
Faulting application path: C:\Program Files\LibHwMon\LibreHardwareMonitor.exe
Faulting module path: C:\Windows\System32\KERNELBASE.dll
Report Id: ffcd2f2d-3df8-418a-b5c0-205e5ad772cb
Faulting package full name: 
Faulting package-relative application ID: 

Application: LibreHardwareMonitor.exe
Framework Version: v4.0.30319
Description: The process was terminated due to an unhandled exception.
Exception Info: System.InvalidOperationException
   at System.ThrowHelper.ThrowInvalidOperationException(System.ExceptionResource)
   at System.Collections.Generic.List`1+Enumerator[[System.__Canon, mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089]].MoveNextRare()
   at LibreHardwareMonitor.UI.TreeModel+<GetChildren>d__7.MoveNext()
   at Aga.Controls.Tree.TreeViewAdv.ReadChilds(Aga.Controls.Tree.TreeNodeAdv, Boolean)
   at Aga.Controls.Tree.TreeViewAdv._model_StructureChanged(System.Object, Aga.Controls.Tree.TreePathEventArgs)
   at LibreHardwareMonitor.UI.Node+NodeCollection.InsertItem(Int32, LibreHardwareMonitor.UI.Node)
   at LibreHardwareMonitor.UI.MainForm.SubHardwareAdded(LibreHardwareMonitor.Hardware.IHardware, LibreHardwareMonitor.UI.Node)
   at LibreHardwareMonitor.UI.MainForm.HardwareAdded(LibreHardwareMonitor.Hardware.IHardware)
   at LibreHardwareMonitor.Hardware.HardwareEventHandler.Invoke(LibreHardwareMonitor.Hardware.IHardware)
   at LibreHardwareMonitor.Hardware.Network.NetworkGroup.UpdateNetworkInterfaces(LibreHardwareMonitor.Hardware.ISettings)
   at System.Threading.ExecutionContext.RunInternal(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object, Boolean)
   at System.Threading.ExecutionContext.Run(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object, Boolean)
   at System.Threading.ExecutionContext.Run(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object)
   at System.Net.NetworkInformation.NetworkChange+AddressChangeListener.AddressChangedCallback(System.Object, Boolean)
   at System.Threading._ThreadPoolWaitOrTimerCallback.PerformWaitOrTimerCallback(System.Object, Boolean)

Zlikster, 2021-08-28:
Have you had a chance to look on this issue? Still same problem with 0.8.8 ver.

iiinsanitea, 2021-11-28:
This is also happening for me as well. Version 0.8.8.

Deviqx, 2023-04-13:
My issue was #1038 .  I was able to finally stop the error from occuring by turning off any temps in the systray. Only the program icon now sits in the systray and I've also noticed after a recent windows update that multiple instances of temperatures in the systray were occurning after a sleep cycle.