LibreHardwareMonitor/LibreHardwareMonitor #472: Some HDD Sensors disappearing in ci264 Regression

Eno77, 2021-06-16:
I Updated to ci264 (actual build) today from ci172

Some of the SATA HDD Sensor Readings disappear some minutes after start (initially everything is OK) now.
Only "Temperature" and "Used Space" remain. On SSD "Total Bytes Written" Remain too.
Write Activities / Total Activity Read Rate / Write Rate disappear.

M.2 NVME Disk not affected.

New (ci264 after some minutes) vs old (ci 172):
[image]

PhyxionNL, 2021-06-17:
I have increased the 2.5s to 5s in this build: https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/actions/runs/945687340
Can you give it a try?

Eno77, 2021-06-17:
Runs now since a little bit more than 2 hours and all readings remain visible, while in the previous (2,5s?) they disappeared after a couple of minutes.
Seems to be good to me now.

PhyxionNL, 2021-06-17:
Thanks for confirming it, merged!

Eno77, 2021-06-17:
unfortunately after some more hours runtime, I must report:

Read/Write Activity and Read/Write Rates disappeared for M.2 and SATA HDD again with latest Build.
The values for SATA SSD (on the same controller like the HDD) are still visible at the moment.

PhyxionNL, 2021-06-17:
Do you have Visual Studio installed? Maybe you can try some different timeouts and see how much time is required.

Eno77, 2021-06-17:
I can give it a try, but it may take a while (short holidays from tomorrow on)

We are sure, its the timeouts?

I dont know anything about WMI (yet) - and it makes me suspicious that while the values are visible, they are changing with the (1s) polling interval, but may still timeout at 5s? And that only these 4 values are affected is maybe a hint too?

Could you give me the path and line number to the timeout setting? Whe I am back next week, I will clone and do some builds with raised timeouts and run tests for some days.

PhyxionNL, 2021-06-17:
These sensors are pulled from WMI and the timeout is the only thing that changed between ci172 and ci264, so it's likely the problem. 5 seconds to poll some data feels like enough, but you can make some changes here:
https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/blob/master/LibreHardwareMonitorLib/Hardware/Storage/AbstractStorage.cs#L126
It might also be related to https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/blob/master/LibreHardwareMonitorLib/Hardware/Storage/AbstractStorage.cs#L186 if the timeout gives the same HResult.

Eno77, 2021-06-21:
OK, I did some q&d (quick and dirty) measurements from a couple of hours today, will let it run to see if some higher spikes occur:

_21.06.2021 12:03:21 driveIndex: 1 new max Milliseconds: 196
21.06.2021 12:03:21 driveIndex: 2 new max Milliseconds: 14
21.06.2021 12:03:21 driveIndex: 0 new max Milliseconds: 11
21.06.2021 12:03:22 driveIndex: 0 new max Milliseconds: 12
21.06.2021 12:07:39 driveIndex: 1 new max Milliseconds: 4600
21.06.2021 12:23:45 driveIndex: 1 new max Milliseconds: 4642
21.06.2021 12:39:50 driveIndex: 1 new max Milliseconds: 4678
21.06.2021 13:18:09 driveIndex: 2 new max Milliseconds: 31
21.06.2021 16:08:58 driveIndex: 1 new max Milliseconds: 4710
21.06.2021 16:25:04 driveIndex: 1 new max Milliseconds: 4720
21.06.2021 17:29:24 driveIndex: 1 new max Milliseconds: 4727
21.06.2021 17:45:29 driveIndex: 1 new max Milliseconds: 4753
21.06.2021 19:05:54 driveIndex: 1 new max Milliseconds: 4779
21.06.2021 19:21:59 driveIndex: 1 new max Milliseconds: **4787**
21.06.2021 19:29:37 driveIndex: 2 new max Milliseconds: 135
21.06.2021 19:29:37 driveIndex: 0 new max Milliseconds: 35
21.06.2021 19:29:38 driveIndex: 0 new max Milliseconds: 70
21.06.2021 19:30:53 driveIndex: 2 new max Milliseconds: **326**
21.06.2021 19:30:53 driveIndex: 0 new max Milliseconds: **185**_

For what I see I think the catched HRESULT is only thrown on timeouts - and then the 4 Sensors are deactivated.
My intuition says, that if the Query was successful one time, we should not deactivate them later just because the timeout occurred.

A short research brought no hints for a "reliable" timeout - something not smaller then 10s to 20s should be safe.

But like I said, the deactivation of known working readings because of a timeout which is known to have very high spikes sometimes... I dont like it very much. ;)

PhyxionNL, 2021-06-21:
Strange that it's also returning -2146233087 as HResult there. I think we can try to add something like if it fails X times then stop, otherwise try again.

PhyxionNL, 2021-06-22:
@Eno77 Can you give this one a try:
https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/pull/481
I think 10 retries should be enough, but if needed it can be increased.

Eno77, 2021-06-22:
Thanks, I will test it. (I think too, 10 retries should be OK to circumvent deactivation of Readings)

For now until now the new maximum query times for the drives are (continued log from yesterday):

_21.06.2021 19:30:53 driveIndex: 0 new max Milliseconds: 185
21.06.2021 19:38:04 driveIndex: 1 new max Milliseconds: 4789
21.06.2021 19:54:10 driveIndex: 1 new max Milliseconds: 4906
21.06.2021 20:10:15 driveIndex: 1 new max Milliseconds: 4940
21.06.2021 22:18:59 driveIndex: 2 new max Milliseconds: 4767
22.06.2021 05:33:19 driveIndex: 0 new max Milliseconds: 4624_

So after ~24h it looks like around 5s is the max Query time on my System regardless of the drive with the mechanical one (driveID 1) being the quickest to reach the higher numbers.

Eno77, 2021-06-22:
Some new max values occurred today:

_22.06.2021 05:33:19 driveIndex: 0 new max Milliseconds: 4624
22.06.2021 16:32:46 driveIndex: 1 new max Milliseconds: 5121
22.06.2021 16:48:52 driveIndex: 1 new max Milliseconds: 5491
22.06.2021 17:04:58 driveIndex: 1 new max Milliseconds: 5681
22.06.2021 17:37:11 driveIndex: 1 new max Milliseconds: 5864
22.06.2021 17:53:17 driveIndex: 1 new max Milliseconds: 5950_

From tomorrow on I will test the new build

Eno77, 2021-06-23:
OK, ~12 hours now and all readings are stilll there.

I am confident, we can assume there will be no case in which the WMI call takes more than the timeout in a number of consecutive cases.

Maybe raising the timeout to 10 seconds would add an additional buffer.
You may decide because I dont know, if we want this to return as soon as possible, or if it does not matter.

Thanks for the effort!

PhyxionNL, 2021-06-23:
Thanks, based on your data I think it would be wise to raise it a bit more, I think 7.5s would be fine.

commit b8288cbb (2021-06-17): Fix #472
commit 65612752 (2021-06-23): Retry WMI in AbstractStorage (#481)

Fixes #472.
closed 2021-06-23 by PhyxionNL