LibreHardwareMonitor/LibreHardwareMonitor #476: Radeon RX 6700 XT GPU Memory Junction Temperature sensor

benoe77, 2021-06-19:
Hi, could you please add the ram temperature sensor data of the new radeon RX 6700 XT? On 5700 it was correctly displayed, but after I changed the card, this information is missing.

Please let me know if I can help somehow adding it

Btw I'm using Remi Mercier's Fan Control, but as I understand, sensor data comes from LibreHardwareMonitor.

zeule, 2021-06-20:
LHM uses AMD Display Library (ADL) to query information from AMD GPUs and should already display each available sensor. Does the AMD Wattman show the RAM temperature for your card?

benoe77, 2021-06-20:
Unoftunately not, but for example hwinfo does. My request is based on my experience with my previous AMD 5700 card, where this sensor data was available.

PhyxionNL, 2021-06-20:
> 
> 
> Unoftunately not, but for example hwinfo does. My request is based on my experience with my previous AMD 5700 card, where this sensor data was available.

Not all cards have the same sensors 😊

benoe77, 2021-06-20:
I guess the sensor is there, you just cannot read it for now.

zeule, 2021-06-20:
hwinfo reads many sensors for the GPU on my machine too, and the Linux amgdpu driver reads memory temperature too. Obviously, there is another way to access these data than the ADL library.

doveppp, 2025-03-03:
same question

any update?