LibreHardwareMonitor/LibreHardwareMonitor #483: Support for VRAM usage reporting.

MartBlur, 2021-06-28:
Hi, thanks for all your work on this. I just wanted to make a suggestion.

Title should be self-explanatory. Other apps like hwinfo, afterburner and even task manager have full access to this, so I'm surprised it wasn't present here given it's one of the most important metrics I needed monitored, at least as a percentage.

PhyxionNL, 2021-06-29:
Which GPU do you have? A load sensor for GPU Memory (= VRAM) already exists for NVIDIA and AMD cards.

MartBlur, 2021-06-30:
I have an rx480 8gb variant.
The only vram parameter shown is frequency. LHM is 0.8.7.

PhyxionNL, 2021-06-30:
What you're looking for should be in the latest build:

PhyxionNL, 2021-06-30:
What you're looking for should be in the latest build:
https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/actions/runs/964546193

MartBlur, 2021-06-30:
I tested and now there's a lot more card telemetry but memory load still isn't there.

PhyxionNL, 2021-06-30:
I suspect that card doesn't use Overdrive8 (which reports it).

MartBlur, 2021-06-30:
I have a correction: The memory consumption does appear, in MB, now just the _percentage_ measurement is the one missing.