LibreHardwareMonitor/LibreHardwareMonitor #497: Can't control GPU fan speed

dMailonG, 2021-07-21:
I'm unable to control the GPU fan speed using LibreHardwareMonitor, doesn't matter whether it's set to Auto or Manual in the Radeon Software.

Radeon™ RX 5600 XT WINDFORCE OC 6G (rev. 2.0)
GV-R56XTWF2OC-6GD
VBIOS: 017.001.000.068.000000
Driver: 21.6.1

Shahnewaz1996, 2021-07-29:
On the same boat. MSI RX 5700 XT GAMING X, same driver.
Might be related to https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/276. Controlling through Afterburner works.
Global Tuning Settings: Automatic, Default

johny-mnemonic, 2022-04-24:
Same here with PowerColor RX 6800 Red Dragon.
No matter what speed I set it doesn't change the fan speed of my GPU.

exeus, 2022-04-27:
Same here, Asrock RX 5700 XT Taichi, fan shows as 0 rpm can't be controlled, but can be controlled via Afterburner, Asrock Tweak etc.

Arn-Thor, 2022-04-28:
Appear to have same issue: MSI 5700XT Gaming X. MSI Afterburner can read and control fan speeds, but not software relying on LibreHardwareMonitor.