LibreHardwareMonitor/LibreHardwareMonitor #501: Fans not detected on Gigabyte B560 motherboard

kolaqsq, 2021-07-25:
Fans not detected on my B560M AORUS ELITE motherboard
[image]
[image]

kolaqsq, 2021-08-11:
Currently working on motherboard support, managed to show data, but still can't controll fans. Any idea on how to do it?
[image]

WizardTim, 2021-08-18:
What changes did you make to get it working? (I have a Gigabyte X570S Aero G with the same ITE IT8689E)

You can't control fan speeds in Libre Hardware Monitor GUI however there are other applications that use LHM to communicate with hardware, the most recent and maintained https://github.com/Rem0o/FanControl.Releases

kolaqsq, 2021-08-20:
I just searched for every mention of IT8688E chip and wrote a similar code but for IT8689E. You can check it out in my [fork](https://github.com/kolaqsq/LibreHardwareMonitor) (last 2 commits)

As for the FanControl program… Managing my fans with it is my goal in the first place, but unfortunately it still can't do it. The program recognizes fans, but changing pwm value does nothing

[image]
[image]

WizardTim, 2021-08-20:
Thanks for sharing your changes, I'm also trying to control my fans in Fan Control.

If I get the time this weekend I will see if I can replicate your working RPM reporting but unable to set PWMs on my motherboard.

Troubleshooting things I can think of are:
* Have you tried running it with Admin privileges? (once solved my problems with SMBus RGB software)
* Have you tried all fans not just CPU and it doesn't work? (Rules out register address off by one)
* Are fan speeds still controlled by the motherboard based on temperature or are they frozen at last value? (indicates if control switched from BIOS curves to manual OS software)

kolaqsq, 2021-08-20:
Thank you for feedback!

I've already tried all the things you questioned and I have some bad news:
- Running as Admin changes nothing
- All of my fans are uncontrollable
- Control remains in BIOS mode 

In other words it seems that in this chip model fan controls are somewhat changed from previous, making it incompatible with program

WizardTim, 2021-08-20:
I wouldn't say it's bad news yet, pretty confident the chip is almost identical, based on it remaining in BIOS mode for all fans `FAN_MAIN_CTRL_REG` is looking rather suspect to me.

Is there anything interesting like error messages in the Fan Control log file? Skimming through the code it looks like it might return some errors if it knew it was still under BIOS control when it tries to apply new PWM values.

EzraBrooks, 2021-11-11:
I just got a Gigabyte Z6890I Aorus Ultra and it has this same chip and thus same issue. Did either of you find anything more on this since your last updates here?

WizardTim, 2021-11-12:
No sorry, I was able to get kolaqsq's changes running and see fan speeds in Fan Control (now in main branch), I guessed and made a couple small changes to it to try to get control working but nothing worked but I never setup a debug environment so wasn't able to properly test the `FAN_MAIN_CTRL_REG` idea. I didn't spend very much time on it as without a proper datasheet/register map for the chip I don't really know what to focus on.

I ended up just giving up and fine tuning my BIOS fan control curves.

EzraBrooks, 2021-11-12:
Okay, thanks. I might take a poke at this, though I'm not much of a hardware guy. My build is a very densely packed SFF system and relies upon being able to control system fans based on GPU temperature - something that the Gigabyte Smart Fan controls in BIOS/SIV don't allow. Although for the time being I might just see if I can squeeze a third-party fan controller into a side panel somewhere in my case, as much of a mess as that will create...

WizardTim, 2023-01-08:
Update for everyone following this issue. Fan Control now works with this chipset (IT8689E) at least for my motherboard (Gigabyte X570S Aero G). #672 appears to be what added the fan control support, thank you to everyone involved in adding this.

Been testing it for a couple weeks now, everything seems to work perfectly. (Unused fans hidden in screenshot but they all detect along with the PCB temperature sensors and thermistors)
[image]

PhyxionNL, 2023-03-21:
Seems fixed based on comment, closing.

closed 2023-03-21 by PhyxionNL