LibreHardwareMonitor/LibreHardwareMonitor #504: Suspected memory leak

WhAtEvErYoUmEaN, 2021-07-31:
I haven't put any effort into debugging it, but if you leave LibreHardwareMonitor running (using the widget in my case) the allocated virtual memory keeps climbing up into multiple gigabytes

nangusa, 2021-08-02:
Take a look at this: https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/450

WhAtEvErYoUmEaN, 2021-08-04:
@nangusa so you're saying this is either the nVidia driver or the sensor value history? Did i get that correct?
Pardon the question, but i don't see any option to turn the latter off.

I resented to launching LibreHardwareMonitor via the task scheduler and killing/restarting the instance every so often as a workaround in the meantime

nangusa, 2021-08-04:
@WhAtEvErYoUmEaN Yes, that's what I suspect it's your memory leak. In LHM, you can navigate to "Options/Sensor Values Time Window" and set it to 30 seconds to minimize the memory footprint.

If you have an nVidia card, update the drivers to any 47x.xx version.

1usmus, 2021-08-19:
> @WhAtEvErYoUmEaN Yes, that's what I suspect it's your memory leak. In LHM, you can navigate to "Options/Sensor Values Time Window" and set it to 30 seconds to minimize the memory footprint.
> 
> If you have an nVidia card, update the drivers to any 47x.xx version.

**Sensor.cs** 
private TimeSpan _valuesTimeWindow = TimeSpan.Zero;

Unfortunately, even this does not solve the problem. AMD graphics cards...

closed 2025-11-18 by PhyxionNL