LibreHardwareMonitor/LibreHardwareMonitor #505: MSI X570: NUC126 Driver

Bondarz, 2021-08-03:
Hello,

I would like to [use FanControl to control a fan I have installed](https://github.com/Rem0o/FanControl.Releases/issues/309). This can be set in the UEFI, but not in FanControl. Think its because of the NUX126 Driver.

Can I help you with some data?

Mainboard: MSI X570 Unify

Thanks

[image]

sunrunner20, 2021-10-05:
I'd also like this, but could use some help getting my feet wet developing the driver. I have no idea where to start.

BCB57, 2022-04-26:
Same issue: MSI X570 Unify.  East enough to work this one fan through the UEFI, but will follow for a future solution.

SamboyCoding, 2022-04-26:
Just noticed this myself too, on a MSI MEG X570 ACE, the SYS_FAN5 pins are not controllable from LHWM. SYS_FAN1-4 are, and appear as Fans 2-6 in the software, and the Southbridge/Chipset fan is also controllable, and appears as Fan 7, but this one fan is missing.

berniyh, 2023-03-06:
My B550 Gaming Carbon also has such a chip.

Is it at least known (e.g. from hwinfo) by which bus this chip can be accessed?
It's not via EC, otherwise I would've found it already, I guess.

If you search the internet, you find e.g. this:
https://www.nuvoton.com/products/microcontrollers/arm-cortex-m0-mcus/nuc126-usb-series/

But I'm not sure if it's the correct one, since I'd expect to find such a USB device easily, but it doesn't seem to show up in lsusb.
So, is it I2C, or GPIO maybe?
How would you find out? :)

wlkmanist, 2024-06-19:
> But So, is it I2C, or GPIO maybe? How would you find out? :)

Have not yet seen this MS-7C35 board yet, but i assume that SMBus used because of my experience with MS-7751 that also uses additional smbus chip for fans. May be we need just listen to some data on system i2c via logic. I will try that when I receive that board.

wlkmanist, 2024-06-19:
Yep, looks like it's connected via smbus

[image]
[image]