LibreHardwareMonitor/LibreHardwareMonitor #516: [BUG] NO VIDEO CARD TEMPS DETECTED - AMD 5700XT

Klezkk, 2021-08-16:
No video card temperature is detected using both the app and the lib. Running the last stable WHQL drivers from AMD.
Temp sensors areworking just fine with other libraries and monitoring apps.

PhyxionNL, 2021-08-16:
Please add a report here (in app: File > Save Report).

Klezkk, 2021-08-16:
Sure, there you go

[LibreHardwareMonitor.Report.txt](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/files/6993605/LibreHardwareMonitor.Report.txt)

PhyxionNL, 2021-08-16:
Thanks, everything is reported by ADL as ADL_ERR or ADL_ERR_NOT_SUPPORTED. I don't have an AMD GPU myself to try so someone will need to debug it and find out why it's not working.

One thing is a bit odd though, you have 8 identical GPUs reported by ADL. Can you try the attached zip and attach a new report?
[net452.zip](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/files/6993799/net452.zip)
I'm curious if one of these GPUs reports the temps correctly.

PhyxionNL, 2021-08-16:
How many GPUs do you now see @ LHM?

Klezkk, 2021-08-16:
[image]

I have a main one, and program correctly identifies temps and fan speed (in this version you made me download) and many more probing just that 1 param.

Klezkk, 2021-08-16:
https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/pull/427 
This one

PhyxionNL, 2021-08-16:
Thanks, I have expected as much. I guess we must find a way to detect these fake cards in ADL. I will analyze the data and see if there's something possible with it (or with another ADL function).

Klezkk, 2021-08-16:
Here's what the publicly available one says, for reference.
No real developer here, but couldn't you just discard every "fake card" not showing up its relative temperature and fan speed?
[image]

PhyxionNL, 2021-08-17:
Can you give this version a try @Klezkk:
[net452.zip](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/files/6998987/net452.zip)
If there are multiple devices on the same bus it should pick the one with the most available sensors.

Klezkk, 2021-08-17:
Nope, still same output as the last official release
[image]

PhyxionNL, 2021-08-18:
It's because of the D3D sensors I suspect, they somehow don't register for your other devices. Can you attach the log of this version:
[net452.zip](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/files/7009065/net452.zip)

It doesn't change anything but logging.

Klezkk, 2021-08-18:
Attachments are not working, so here's the dump.

`
LibreHardwareMonitor Report

--------------------------------------------------------------------------------

Version: 0.8.8.0

--------------------------------------------------------------------------------

Common Language Runtime: 4.0.30319.42000
Operating System: Microsoft Windows NT 10.0.19043.0
Process Type: 64-Bit

--------------------------------------------------------------------------------

Sensors

|
+- ASRock B550 Phantom Gaming 4 (/motherboard)
|  |
|  +- Nuvoton NCT6796D-R (/lpc/nct6796dr)
|  |  +- Vcore          :      2.4     1.68      2.4 (/lpc/nct6796dr/voltage/0)
|  |  +- Voltage #2     :    1.688    1.688    1.696 (/lpc/nct6796dr/voltage/1)
|  |  +- AVCC           :    3.408    3.408    3.408 (/lpc/nct6796dr/voltage/2)
|  |  +- +3.3V          :     3.36    3.344     3.36 (/lpc/nct6796dr/voltage/3)
|  |  +- Voltage #5     :    1.824    1.824    1.824 (/lpc/nct6796dr/voltage/4)
|  |  +- Voltage #6     :    1.088    1.088    1.096 (/lpc/nct6796dr/voltage/5)
|  |  +- Voltage #7     :    1.352    1.344    1.352 (/lpc/nct6796dr/voltage/6)
|  |  +- 3VSB           :    3.408    3.408    3.408 (/lpc/nct6796dr/voltage/7)
|  |  +- VTT            :    0.912    0.912    0.912 (/lpc/nct6796dr/voltage/9)
|  |  +- Voltage #11    :    1.024    1.024    1.024 (/lpc/nct6796dr/voltage/10)
|  |  +- Voltage #12    :    0.632    0.632    0.632 (/lpc/nct6796dr/voltage/11)
|  |  +- Voltage #13    :    1.088     1.08    1.088 (/lpc/nct6796dr/voltage/12)
|  |  +- Voltage #14    :    0.936    0.936    0.936 (/lpc/nct6796dr/voltage/13)
|  |  +- Voltage #15    :    0.912    0.912    0.912 (/lpc/nct6796dr/voltage/14)
|  |  +- Temperature #1 :       43       43       43 (/lpc/nct6796dr/temperature/1)
|  |  +- Temperature #2 :       39       39       39 (/lpc/nct6796dr/temperature/2)
|  |  +- Temperature #3 :        8        8        8 (/lpc/nct6796dr/temperature/3)
|  |  +- Temperature #5 :       12       12       12 (/lpc/nct6796dr/temperature/5)
|  |  +- Temperature #6 :       31       31       31 (/lpc/nct6796dr/temperature/6)
|  |  +- Fan #1         :  916.497  912.162    937.5 (/lpc/nct6796dr/fan/0)
|  |  +- Fan #2         :  1091.35  1090.47  1094.89 (/lpc/nct6796dr/fan/1)
|  |  +- Fan #3         :  1491.71  1481.89  1501.67 (/lpc/nct6796dr/fan/2)
|  |  +- Fan #4         :  1415.09  1393.19  1424.05 (/lpc/nct6796dr/fan/3)
|  |  +- Fan #5         :        0        0        0 (/lpc/nct6796dr/fan/4)
|  |  +- Fan #6         :        0        0        0 (/lpc/nct6796dr/fan/5)
|  |  +- Fan #7         :        0        0        0 (/lpc/nct6796dr/fan/6)
|  |  +- Fan Control #1 :      100      100      100 (/lpc/nct6796dr/control/0)
|  |  +- Fan Control #2 :  29.8039  29.8039  29.8039 (/lpc/nct6796dr/control/1)
|  |  +- Fan Control #3 :      100      100      100 (/lpc/nct6796dr/control/2)
|  |  +- Fan Control #4 :      100      100      100 (/lpc/nct6796dr/control/3)
|  |  +- Fan Control #5 :  30.1961  30.1961  30.1961 (/lpc/nct6796dr/control/4)
|  |  +- Fan Control #6 :  30.1961  30.1961  30.1961 (/lpc/nct6796dr/control/5)
|  |  +- Fan Control #7 :  30.1961  30.1961  30.1961 (/lpc/nct6796dr/control/6)
|
+- AMD Ryzen 5 3600 (/amdcpu/0)
|  +- Core (SVI2 TFN) :  1.19375   1.1875      1.2 (/amdcpu/0/voltage/0)
|  +- SoC (SVI2 TFN) :   1.0875   1.0875   1.0875 (/amdcpu/0/voltage/1)
|  +- Core #1 VID    :      1.2   1.1875      1.2 (/amdcpu/0/voltage/2)
|  +- Core #2 VID    :      1.2   1.1875      1.2 (/amdcpu/0/voltage/3)
|  +- Core #3 VID    :      1.2   1.1875      1.2 (/amdcpu/0/voltage/4)
|  +- Core #4 VID    :      1.2   1.1875      1.2 (/amdcpu/0/voltage/5)
|  +- Core #5 VID    :      1.2   1.1875      1.2 (/amdcpu/0/voltage/6)
|  +- Core #6 VID    :      1.2   1.1875      1.2 (/amdcpu/0/voltage/7)
|  +- TDC            :  7.30752  1.13549  14.1175 (/amdcpu/0/current/0)
|  +- EDC            :       90       90       90 (/amdcpu/0/current/1)
|  +- Package        :  34.1498  24.9801  41.0762 (/amdcpu/0/power/0)
|  +- Core #1 (SMU)  :  1.11493 0.277949  2.55426 (/amdcpu/0/power/1)
[... 2529 lines cut ...]
--------------------------------------------------------------------------------

Realtek PCIe GbE Family Controller
Up

Data Uploaded
0,02774381

Data Downloaded
0,3134917

Upload Speed
0

Download Speed
0

Network Utilization
0

--------------------------------------------------------------------------------

Corsair HXi/RMi series PSU Hardware

`

Klezkk, 2021-08-18:
[LibreHardwareMonitor.Report.txt](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/files/7009738/LibreHardwareMonitor.Report.txt)

PhyxionNL, 2021-08-18:
Thank you, can you give this version a try:
[net452.zip](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/files/7010130/net452.zip)

Klezkk, 2021-08-18:
[image]
[LibreHardwareMonitor.Report.txt](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/files/7010151/LibreHardwareMonitor.Report.txt)

PhyxionNL, 2021-08-19:
A new version to try 🙂
[net452.zip](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/files/7012582/net452.zip)

Klezkk, 2021-08-19:
Seems ok! 
[LibreHardwareMonitor.Report.txt](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/files/7015418/LibreHardwareMonitor.Report.txt)
[image]

commit 22d4c04a (2021-08-19): Fixes #516
commit e061f722 (2021-08-19): Fixes #516 (#518)
closed 2021-08-19 by PhyxionNL